mvp/uhubctl #83: Doesn't work on Lenovo T-560

pillarsdotnet, 2018-03-27:
```
root@T560:/usr/src/uhubctl# ./uhubctl
No compatible smart hubs detected!
Run with -h to get usage info.
There were permission problems while accessing USB.
To fix this, run this tool as root using 'sudo uhubctl',
or add one or more udev rules like below
to file '/etc/udev/rules.d/52-usb.rules':
SUBSYSTEM=="usb", ATTR{idVendor}=="2001", MODE="0666"
then run 'sudo udevadm trigger --attr-match=subsystem=usb'
root@T560:/usr/src/uhubctl# echo 'SUBSYSTEM=="usb", ATTR{idVendor}=="2001", MODE="0666"' > /etc/udev/rules.d/52-usb.rules
root@T560:/usr/src/uhubctl# udevadm trigger --attr-match=subsystem=usb
root@T560:/usr/src/uhubctl# ./uhubctl
No compatible smart hubs detected!
Run with -h to get usage info.
There were permission problems while accessing USB.
To fix this, run this tool as root using 'sudo uhubctl',
or add one or more udev rules like below
to file '/etc/udev/rules.d/52-usb.rules':
SUBSYSTEM=="usb", ATTR{idVendor}=="2001", MODE="0666"
then run 'sudo udevadm trigger --attr-match=subsystem=usb'
root@T560:/usr/src/uhubctl#
```

pillarsdotnet, 2018-03-27:
[lsusb -v &>lsusb-v.txt](https://github.com/mvp/uhubctl/files/1853804/lsusb-v.txt)

mvp, 2018-03-27:
It looks like your laptop does not have any compatible hardware attached.
Supported devices will show like this in `lsusb -v` output (watch for `Per-port power switching`):
```
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             6
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
```

As for adding udev rules - they are not necessary if you are running as root.
If you want uhubctl to work as non-root, only then you need to add udev rules, and use vendor id of hub you are insterested in.
This error message about permission issue isn't quite correct - it only shows up because you don't have any supported devices (I should fix this misleading error message).

If you obtain any hub on supported list and attach to your laptop, it should work.

closed 2018-03-27 by mvp