mvp/uhubctl #85: Raspi Model B

gruessung, 2018-03-31:
I have trouble to turn off a specific usb port on my raspi model b.
I see that port 1 is usb ethernet.
So port 2 or 3 have to be my LED stripe.
But 
`sudo ./uhubctl  -a off -p 2`
and
`sudo ./uhubctl  -a off -p 3
`didn't work.

Do you have any suggestions?

mvp, 2018-03-31:
Raspberry Pi Model B hardware does not support power switching. You need model B+, 2 B, or 3 B. If you get compatible model, use port 2 to control all ports 2-5. Port 1 controls Ethernet+wifi.

closed 2018-03-31 by mvp

gtzigk, 2019-02-16:
Hello, 
is it possible on Pi 3 Model B to control just one Port from 2-5 separately from others and not all of them?
[image]
Unfortunately I guess not...

Thanks

mvp, 2019-02-16:
Check https://github.com/mvp/uhubctl/issues/138.

gtzigk, 2019-02-16:
Thanks for the response!