mvp/uhubctl #89: uhubctl in my python code on raspberry pi

sameenashaikh29, 2018-04-05:
Hi,
I followed the steps to install uhubctl on my raspberry pi. 
sudo ./uhubctl a 0 p 2 does power off all my ports
sudo ./uhubctl a 1 p 2 does power on all my ports
I dont' have support for individual ports through uhubctl. But that's not a concern as of now.
When i use the command in my python code where i need enabling nd disabling the power with the instruction os.system("sudo ./uhubctl a 0 p 2") and os.system("sudo ./uhubctl a 1 p 2")  at different instances in my code, the power off does happen but the power back on doesn't. The code is unstable due to these commands only.. Would appreciate if you could help with this
Hoping to get a quick response

Thanks & Regards
Sameena Shaikh

sameenashaikh29, 2018-04-05:
Hi,
Ganged switching on and off is fine but how can i use it in my python code. Not as a command.
I've tried with these instructions
 os.system("sudo ./uhubctl -a 0 -p 2") and os.system("sudo ./uhubctl -a 1 -p 2")
in my python code, but it doesn't seem to be stable.
Through commands they work fine, but in my code it does power off, clicks images and for me to access those images the power to usb should be enabled, but that isn't happening. Or it happens sometimes and sometimes doesn't.
sudo ./uhubctl -a 0 -p 2
sudo ./uhubctl -a 1 -p 2
and with this command i'm using port 2 itself right and not port 1?

Thanks & Regards

sameenashaikh29, 2018-04-05:
Thank you for your previous response

mvp, 2018-04-05:
You need to do 2 things: 

1. Copy/install uhubctl executable in some standard system path, e.g. /usr/bin, /usr/sbin or /usr/local/bin, so you don't have to use `./`.
2. You need to configure udev USB permissions, so you don't need to use sudo.

After that, you will be able to invoke uhubctl just by name, even in your python code, and won't need to be root or use sudo.

closed 2018-04-05 by mvp