mvp/uhubctl #90: Setting port off causes port power to cycle on-off-on (rather than stay off)

mypiandrew, 2018-04-10:
Using a custom Rasberry Pi board with LAN9512 device,  As per the reference design we have used a MIC2026 device to gate the +5V power feed to the USB ports, which is controlled by the PRTCTL2/3 lines (see page 17 of LAN9512 chip for details)

When using your neat program I find that setting a port off causes port power to cycle on-off-on (rather than stay off)

What am I missing here ?

```
root@raspberrypi:~/uhubctl-master# ./uhubctl
Current status for hub 1-1.1 [0424:9512, USB 2.00, 3 ports]
  Port 1: 0503 power highspeed enable connect [0424:ec00]
  Port 2: 0100 power
  Port 3: 0503 power highspeed enable connect [0930:6544 TOSHIBA TransMemory 0022CFF653ECC161723533D3]
Current status for hub 1-1.1 [0424:9512, USB 2.00, 3 ports]
  Port 1: 0503 power highspeed enable connect [0424:ec00]
  Port 2: 0100 power
  Port 3: 0503 power highspeed enable connect [0930:6544 TOSHIBA TransMemory 0022CFF653ECC161723533D3]
root@raspberrypi:~/uhubctl-master# ./uhubctl -a 0 -p 3
Current status for hub 1-1.1 [0424:9512, USB 2.00, 3 ports]
  Port 3: 0503 power highspeed enable connect [0930:6544 TOSHIBA TransMemory 0022CFF653ECC161723533D3]
Sent power off request
New status for hub 1-1.1 [0424:9512, USB 2.00, 3 ports]
  Port 3: 0000 off

```
dmesg shows :
```
[ 1604.051219] usb 1-1.1.2: USB disconnect, device number 18
[ 1605.060842] usb 1-1.1.2: new high-speed USB device number 19 using dwc_otg
[ 1605.195741] usb 1-1.1.2: New USB device found, idVendor=0930, idProduct=6544
[ 1605.195754] usb 1-1.1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1605.195763] usb 1-1.1.2: Product: TransMemory
[ 1605.195771] usb 1-1.1.2: Manufacturer: TOSHIBA
[ 1605.195779] usb 1-1.1.2: SerialNumber: 0022CFF653ECC161723533D3
[ 1605.204875] usb-storage 1-1.1.2:1.0: USB Mass Storage device detected
[ 1605.205222] scsi host1: usb-storage 1-1.1.2:1.0
[ 1606.261745] scsi 1:0:0:0: Direct-Access     TOSHIBA  TransMemory           PQ: 0 ANSI: 6
[ 1606.263006] sd 1:0:0:0: Attached scsi generic sg1 type 0
[ 1606.263691] sd 1:0:0:0: [sdb] 60620800 512-byte logical blocks: (31.0 GB/28.9 GiB)
[ 1606.264040] sd 1:0:0:0: [sdb] Write Protect is off
[ 1606.264053] sd 1:0:0:0: [sdb] Mode Sense: 4f 00 00 00
[ 1606.264386] sd 1:0:0:0: [sdb] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[ 1606.269586]  sdb: sdb1
[ 1606.272331] sd 1:0:0:0: [sdb] Attached SCSI removable disk
```

How do I make "no" mean "no" here (i.e. make the power off mean OFF)

mvp, 2018-04-10:
Some device drivers in kernel are surprised by USB device being turned off and automatically try to power it back on. You can use option `-r N` where N is some number from 10 to 100 to fix this - uhubctl will try to turn power off many times in quick succession, and it should suppress that. This may be eventually fixed in kernel, see more discussion here: https://www.spinics.net/lists/linux-usb/msg157413.html

mypiandrew, 2018-04-10:
Hi

Yes adding -r 80 appears to force it :)

mvp, 2018-04-10:
Good, I'm glad you have working solution - this is why I added this option. I wish Linux kernel developers improved support for power control in kernel... BTW, using -r should not be necessary on Mac or FreeBSD.

closed 2018-04-10 by mvp