mvp/uhubctl #91: Double entries in list

jensolsson, 2018-04-10:
Love that it seem to be able to control some USB3 hubs now and I understand that I will see one USB2 and one USB3 device for each. However for me I see two USB3 and two USB2 ports of each. I guess this should not be expected?

I only have one  Apple Watch Magnetic Charging Cable connected.

Kind regards
Jens

Jens-Home-iMac:~ deadlock$ uhubctl -l 20-2.4 -p 3
Current status for hub 20-20.4 [2109:0812 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 3: 02a0 power 5gbps Rx.Detect
Current status for hub 20-2.4 [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 3: 0103 power enable connect [05ac:1392 Apple Inc. Apple Watch Magnetic Charging Cable DLC502709HQFTPG48]
Current status for hub 20-20.4 [2109:0812 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 3: 02a0 power 5gbps Rx.Detect
Current status for hub 20-2.4 [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 3: 0103 power enable connect [05ac:1392 Apple Inc. Apple Watch Magnetic Charging Cable DLC502709HQFTPG48]

mvp, 2018-04-10:
This is issue #33. Make sure you have libusb-1.0.22 installed, and it should start working properly.

mvp, 2018-04-11:
Hmm. That doesn't look right. I wish I could reproduce this. Can you show uhubctl output without using port constraint? I don't quite understand your devices topology. I suspect that this is related to USB3 duality handling. It would be helpful if you also tried to see output with -e, which disables USB3 duality handling.

jensolsson, 2018-04-11:
Sure, here it is without and with -e. I installed it also on an ubuntu box and it works fine there, only on OSX is the problem. Allthough I have different usb hubs on the computers and unfortunately not possible for me to move them at this point.

Jens-Home-iMac:~ deadlock$ uhubctl
Current status for hub 20-20.4 [2109:0812 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0080 off
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 20-20.1 [2109:0812 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 20-2.4 [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0103 power enable connect [05ac:1392 Apple Inc. Apple Watch Magnetic Charging Cable DLC502709HQFTPG48]
  Port 4: 0100 power
Current status for hub 20-2.1 [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0103 power enable connect [1050:0407 Yubico Yubikey 4 OTP+U2F+CCID]
  Port 2: 0100 power
  Port 3: 0103 power enable connect [067b:2303 Prolific Technology Inc. USB-Serial Controller]
  Port 4: 0100 power
Current status for hub 20-20 [2109:0812 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0203 power 5gbps U0 enable connect [2109:0812 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 0203 power 5gbps U0 enable connect [2109:0812 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
Current status for hub 20-2 [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0503 power highspeed enable connect [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
Current status for hub 20-20.4 [2109:0812 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0080 off
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 20-20.1 [2109:0812 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 20-2.4 [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0103 power enable connect [05ac:1392 Apple Inc. Apple Watch Magnetic Charging Cable DLC502709HQFTPG48]
  Port 4: 0100 power
Current status for hub 20-2.1 [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0103 power enable connect [1050:0407 Yubico Yubikey 4 OTP+U2F+CCID]
  Port 2: 0100 power
  Port 3: 0103 power enable connect [067b:2303 Prolific Technology Inc. USB-Serial Controller]
  Port 4: 0100 power
Current status for hub 20-20 [2109:0812 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0203 power 5gbps U0 enable connect [2109:0812 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 0203 power 5gbps U0 enable connect [2109:0812 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
Current status for hub 20-2 [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0503 power highspeed enable connect [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
Jens-Home-iMac:~ deadlock$ uhubctl -e
Current status for hub 20-20.4 [2109:0812 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0080 off
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 20-20.1 [2109:0812 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
[... 30 lines cut ...]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 20-2.4 [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0103 power enable connect [05ac:1392 Apple Inc. Apple Watch Magnetic Charging Cable DLC502709HQFTPG48]
  Port 4: 0100 power
Current status for hub 20-2.1 [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0103 power enable connect [1050:0407 Yubico Yubikey 4 OTP+U2F+CCID]
  Port 2: 0100 power
  Port 3: 0103 power enable connect [067b:2303 Prolific Technology Inc. USB-Serial Controller]
  Port 4: 0100 power
Current status for hub 20-20 [2109:0812 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0203 power 5gbps U0 enable connect [2109:0812 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 0203 power 5gbps U0 enable connect [2109:0812 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
Current status for hub 20-2 [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0503 power highspeed enable connect [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
Jens-Home-iMac:~ deadlock$

mvp, 2018-04-11:
Thanks, -e didn't change anything. I need to find a way to reproduce this. It's almost as if libusb is returning duplicate devices list, but I'm not quite sure. Exactly the same code works fine on Linux, which might manifest libusb bug on OSX. Which version of OSX you have? I have most recent High Sierra on MacBook Pro 2014.

jensolsson, 2018-04-11:
I have High Sierra 10.13.2 on an iMac with 4 USB3 ports

I have a 10 port USB hub connected to one of the iMacs ports and I also have a 7 port USB hub connected to another one of the iMacs ports.

mvp, 2018-04-11:
`uhubctl -e` should simply dump list of all hubs as enumerated by libusb. If that list contains duplicates, it must be libusb bug. We can probably mitigate this by removing returned duplicates by uhubctl usb_find_hubs() - it should fix this, but sounds like hacky way to patch dumb libusb code.

jensolsson, 2018-04-11:
I agree, maybe worth it though... anyway I could simply verify this is the problem? using some other tool? possible to install lsusb on a mac?

jensolsson, 2018-04-11:
brew update && brew tap jlhonora/lsusb && brew install lsusb

I managed to install lsusb using brew and a tap on a mac and I have this output

Jens-Home-iMac:~ deadlock$ lsusb
Bus 020 Device 007: ID 0781:5589 SanDisk Corporation SanDisk Ext SSD  Serial: 164743400252
Bus 020 Device 009: ID 2109:0812 VIA Labs, Inc. USB3.0 Hub 
Bus 020 Device 013: ID 2109:0812 VIA Labs, Inc. USB3.0 Hub 
Bus 020 Device 012: ID 2109:0812 VIA Labs, Inc. USB3.0 Hub 
Bus 020 Device 002: ID 0a5c:4500 Broadcom Corp. BRCM20702 Hub 
Bus 020 Device 006: ID 05ac:828d Apple Inc. Bluetooth USB Host Controller 
Bus 020 Device 003: ID 05ac:8511 Apple Inc. FaceTime HD Camera (Built-in)  Serial: CC2D6E01QGFDYQF0
Bus 020 Device 014: ID 2109:2812 VIA Labs, Inc. USB2.0 Hub 
Bus 020 Device 015: ID 2109:2812 VIA Labs, Inc. USB2.0 Hub 
Bus 020 Device 008: ID 2109:2812 VIA Labs, Inc. USB2.0 Hub 
Bus 020 Device 011: ID 2109:2812 VIA Labs, Inc. USB2.0 Hub 
Bus 020 Device 005: ID 05ac:1392 Apple Inc. Apple Watch Magnetic Charging Cable  Serial: DLC502709HQFTPG48
Bus 020 Device 010: ID 2109:2812 VIA Labs, Inc. USB2.0 Hub 
Bus 020 Device 024: ID 067b:2303 Prolific Technology, Inc. USB-Serial Controller 
Bus 020 Device 025: ID 1050:0407 1050 Yubikey 4 OTP+U2F+CCID 
Bus 000 Device 001: ID 1d6b:LPTH Linux Foundation USB 3.0 Bus

mvp, 2018-04-11:
I don't think lsusb for Mac even exists. In https://github.com/libusb/libusb/examples, you can find and build listdevs. If it shows duplicates, there you have reproducible test case.

But frankly, uhubctl does pretty much the same enumeration, it just filters only hubs, and only hubs which support PPPS. If libusb enumeration is bug-free, uhubctl should never show duplicates.

jensolsson, 2018-04-11:
But seems it does not use libusb....

mvp, 2018-04-11:
Note that this version of lsusb is simply parsing device tree dump by some apple tool - it doesn't use libusb at all. We need one that is actually using libusb.

jensolsson, 2018-04-11:
Here is also some output from OSX built in command which maybe could be interesting.
I realise that I will use uhubctl on linux only so for me this will not be an actual problem, but I am sure other people want to use it on osx

Jens-Home-iMac:~ deadlock$ ioreg -p IOUSB -w 0
+-o Root  <class IORegistryEntry, id 0x100000100, retain 15>
  +-o AppleUSBXHCI Root Hub Simulation@14000000  <class AppleUSBRootHubDevice, id 0x1000002f7, registered, matched, active, busy 0 (19 ms), retain 14>
    +-o BRCM20702 Hub@14600000  <class AppleUSBDevice, id 0x1000002f9, registered, matched, active, busy 0 (21 ms), retain 15>
    | +-o Bluetooth USB Host Controller@14630000  <class AppleUSBDevice, id 0x100000331, registered, matched, active, busy 0 (1505 ms), retain 20>
    +-o FaceTime HD Camera (Built-in)@14500000  <class AppleUSBDevice, id 0x10000030a, registered, matched, active, busy 0 (61813 ms), retain 22>
    +-o SanDisk Ext SSD@14a00000  <class AppleUSBDevice, id 0x100000820, registered, matched, active, busy 0 (19076 ms), retain 14>
    +-o USB2.0 Hub             @14200000  <class AppleUSBDevice, id 0x100000860, registered, matched, active, busy 0 (19 ms), retain 16>
    | +-o USB2.0 Hub             @14210000  <class AppleUSBDevice, id 0x10000088d, registered, matched, active, busy 0 (15 ms), retain 16>
    | | +-o USB-Serial Controller@14213000  <class AppleUSBDevice, id 0x1000009a0, registered, matched, active, busy 0 (17541 ms), retain 17>
    | | +-o Yubikey 4 OTP+U2F+CCID@14211000  <class AppleUSBDevice, id 0x1000009ad, registered, matched, active, busy 0 (48675 ms), retain 24>
    | +-o USB2.0 Hub             @14240000  <class AppleUSBDevice, id 0x1000008a3, registered, matched, active, busy 0 (38 ms), retain 15>
    |   +-o Apple Watch Magnetic Charging Cable@14243000  <class AppleUSBDevice, id 0x100230fe8, registered, matched, active, busy 0 (2562 ms), retain 14>
    +-o USB3.0 Hub             @14900000  <class AppleUSBDevice, id 0x100000876, registered, matched, active, busy 0 (16 ms), retain 16>
    | +-o USB3.0 Hub             @14910000  <class AppleUSBDevice, id 0x1000008ba, registered, matched, active, busy 0 (41 ms), retain 14>
    | +-o USB3.0 Hub             @14940000  <class AppleUSBDevice, id 0x1000008d0, registered, matched, active, busy 0 (35 ms), retain 14>
    +-o USB2.0 Hub             @14300000  <class AppleUSBDevice, id 0x1000008f8, registered, matched, active, busy 0 (17 ms), retain 15>
      +-o USB2.0 Hub             @14310000  <class AppleUSBDevice, id 0x10000090e, registered, matched, active, busy 0 (23 ms), retain 14>

mvp, 2018-04-11:
ioreg output doesn't seem to contain duplicates - which is good, but doesn't say anything about libusb bugs. I will probably add a check to remove hub duplicates if libusb happens to return them. But we also need to file a bug against libusb. I will do this in next few days, hopefully before weekend.

jensolsson, 2018-04-11:
Sounds like a good plan! Should also say I tried out the new version of uhubctl on 7 port AmazonBasics USB3 hub today and it seem to work fine. Will test some more.

mvp, 2018-04-15:
It wasn't libusb after all. Fixed in 2311a1c76.

closed 2018-04-15 by mvp