mvp/uhubctl #93: Ability to switch off and stay off depends on connected device.

adrianboyko, 2018-04-15:
Hi. I've been testing uhubctl with two very old hubs which I believe are actually the same design:
* Belkin F5U014-OE
* SIIG US2228

With both devices, I am able to turn ports off and on. However, whether the port **stays off** depends on what's connected to it. 
* If no device is connected, the port stays off. 
* If a [mVox MV100](https://archive.org/stream/generalmanual_000075268#page/n1/mode/2up) speakerphone is connected, the port stays off (hurray!) 
* Any other device I've tested so far seems to somehow motivate the hub to turn the power back on after it briefly turns off.

Unfortunately, the device I'm actually trying to power off/on is an Adafruit Trinket M0 and it is one of the devices that somehow causes power to turn back on. Any idea why the particular type of device connected makes a difference?

mvp, 2018-04-15:
I believe this is duplicate for #90.

This is Linux kernel issue, and should not be reproducible on other platforms.
Some device drivers in kernel are surprised by USB device being turned off and automatically try to power it back on. You can use option -r N where N is some number from 10 to 100 to fix this - uhubctl will try to turn power off many times in quick succession, and it should suppress that. This may be eventually fixed in kernel, see more discussion here: https://www.spinics.net/lists/linux-usb/msg157413.html

mvp, 2018-04-15:
Thanks for reporting SIIG US2228, I will add it to the list.

closed 2018-04-15 by mvp

mvp, 2018-04-15:
Hmm. These hub models both appear to be extremely old (I found some references dating back to 2001), and it doesn't seem possible to buy them anywhere now, not even eBay. Also they seem to be USB 1.1 only.

Where did you get these hubs? Is is possible to buy them anywhere these days? I don't want to add something that doesn't really exist in the wild.