mvp/uhubctl #94: Anker 4 Port USB 3.0 Model : A7516

leperjulien, 2018-04-18:
Hi,

So i know mine isn't in the supported list but i gave it a try, and that's weird because when i plug the hub to my Macbookpro (with nothing attached to it) and run `uhubctl` i got what i expected : 

```
No compatible smart hubs detected!
Run with -h to get usage info.
```

but if i plug in my iphone to it and run the same command again i got : 

```
Current status for hub 38-4 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect []
```
then i tried `uhubctl -a off -p 4` and my iphone stopped charging !

but then i added a small 5V fan for PC from amazon to the hub (the fan started) but when i tried to do `uhubctl -a off -p 123` the fan kept running :/

`Current status for hub 38-4 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
Sent power off request
New status for hub 38-4 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off`

So, i would only works with phones or is there something wrong with my fan ?

Thanks guys !

leperjulien, 2018-04-19:
Thanks a lot for the (crazy)-fast answer ;)

As you thought i'm experimenting on an old macbook pro which has USB 2.0 only, and i was hoping to use your tool and my anker hub on a raspberry project and since neither my Macbook or my raspberry pi use USB 3.0 i'm afraid i need to look for an other way to achieve this..

```
Warning: libusb 1.0.22 is already installed and up-to-date
To reinstall 1.0.22, run `brew reinstall libusb`

```

I will try it anyway and repeat same test to let you know if this Hub can be added to the list , once i have access to a computer using USB 3.0 :)

Thanks for the assist!

mvp, 2018-04-19:
Note that Raspberry Pi internal USB hub is supported by uhubctl. The only catch is that you can control only port 2, and it will turn off/on ports 2-5 at once.

Also, you can attach any hub on supported list to Raspberry Pi and control ports individually.

leperjulien, 2018-04-28:
Yep i saw this but since i have 3 devices plugged in usb and i need to control them individually :)
Thats why i opted for a D link hub and purchased on amazon but i don't know if the one they sell is a rev C!

Will update here before monday!

mvp, 2018-04-28:
Yes, unfortunately, D-Link makes it very difficult to tell those models apart. I recommend looking very carefully at pictures for your hub on Amazon or eBay. Sometimes they have a picture of serial number, model name and hw revision.

leperjulien, 2018-04-28:
Awesome !! Just got my DUB-H4 from amazon and uhubctl is working perfectly :D 

If anyone is wondering which one is compatible , This one is a rev D1 : 

[](https://www.amazon.fr/gp/product/B00006B7DA/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1)

Thanks a lot for the help :-)

mvp, 2018-04-28:
Glad to hear that it works for you. DUB-H4 rev D1 is already in supported list, and we know that it works well - but still thanks for extra confirmation. Closing this issue...

closed 2018-04-28 by mvp