mvp/uhubctl #95: AmazonBasics USB3 hub

jensolsson, 2018-04-19:
I have three 7-port AmazonBasics USB3 hubs that I bought for using with uhubctl. I cannot determine if they work or not though.

The devices certainly disconnect from the computer, but the question is if they still charge.
My current theory is that iphones stop charging while android devices continue to charge. Could it be that iPhones automatically stop charging at the same time the data connection is dropped?

mvp, 2018-04-19:
I think the only reasonable test would be using some really dumb device like USB fan or light to see if it stops spinning or shining.

Or, better yet, use voltmeter on USB wire to see if VBUS is gone...

jensolsson, 2018-04-25:
> @jensollson, can you try connecting USB fan or USB light and see if they stop working? There are also USB cables with LED light which could be used for this purpose.

I just bought something called a usb multimeter that meassures ampere and voltage. Turned out I was using an incorrect switch. Everything works now!

Kind regards
Jens

Skickat från min iPhone

mvp, 2018-04-25:
Ah, that's awesome! At last, we have final confirmation that it actually turns off - thank you! I will close this issue then.

closed 2018-04-25 by mvp