mvp/uhubctl #96: Power comes back on after ~8s on RaspberryPi3

lademeister, 2018-04-23:
I am using a HDMI Display with usb touch that is powered by usb as well.
It has no ability to switch off its backlight, so I tried uhubctl.
Power down of usb works but after roughly 8s the USB ports power on again.
I used the command 

>  sudo ./uhubctl -a 2 -p 2

This also happens when I set 
 
> vcgencmd display_power 0

Why does it awake again? I also tried -r option, no change.
The only other usb device is a wireless keyboard receiver.

mvp, 2018-04-23:
This appears to be duplicate for issue #90.

Some device drivers in kernel are surprised by USB device being turned off and automatically try to power it back on. You can use option -r N where N is some number from 10 to 100 to fix this - uhubctl will try to turn power off many times in quick succession, and it should suppress that. If you tried option -r and it didn't help, try using larger number for -r, 100 almost always works. This may be eventually fixed in kernel, see more discussion here: https://www.spinics.net/lists/linux-usb/msg157413.html

mvp, 2018-04-23:
One more thing: -a 2 cycles power. If you want to turn power off, use -a 0.

lademeister, 2018-04-23:
That’s it! Thanks for pointing me in the right direction.
I only tried -r 1 up to 3.
100 doesn’t work for me either but 500 did.
Great answer - I didn’t find that anywhere else. Thanks!

closed 2018-04-23 by lademeister

mvp, 2018-04-23:
Glad to hear it helped you. Perhaps uhubctl should have an option to keep trying turn off automatically until it is actually off.

lackdaz, 2018-09-21:
> Some device drivers in kernel are surprised by USB device being turned off and automatically try to power it back on. You can use option -r N where N is some number from 10 to 100 to fix this - [...]

> The interaction is a bit more subtle. I suspect it's a result of the Ethernet controller being connected via back-to-back UTMI or HSIC rather than USB signals.
> 
> If I disable "power" to the port, no messages are generated in dmesg. Pings to hosts fail with a timeout. If I subsequently re-enable the port power, the Ethernet controller disconnects and then immediately reconnects. Pings start working again for me.

I can confirm that this works for fixing any power off -> cycling with the Raspberry Pi 3B+ as well -  although its such an odd fix.