mvp/uhubctl #98: Release versions build with a git error

severach, 2018-05-14:
````
fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
````

The release version doesn't have any git information. A line in `Makefile` needs more quotes. I thought quote escaping was an error until I looked at the code. The extra quotes allows spaces in the version.

````
--- Makefile.orig	2018-05-14 14:25:05.722009243 -0400
+++ Makefile	2018-05-14 14:25:05.725342626 -0400
@@ -16,5 +16,5 @@
 CFLAGS += -Wall -Wextra -std=c99 -pedantic
 GIT_VERSION := $(shell git describe --abbrev=4 --dirty --always --tags)
-CFLAGS += -DPROGRAM_VERSION=\"$(GIT_VERSION)\"
+CFLAGS += -DPROGRAM_VERSION="\"$(GIT_VERSION)\""
 
 ifeq ($(UNAME_S),Linux)
````

mvp, 2018-05-14:
I am not sure how to make release version to carry version information without embedding it in makefile or some header - this is what I had before, and it was very awkward to support.

As for having spaces in version, I think it's not desirable. Version string should follow standard semantic versioning convention, and shouldn't have any spaces.

severach, 2018-05-14:
I fixed it [with](https://aur.archlinux.org/packages/uhubctl/)

````
sed -e 's/^GIT_VERSION :=/#&/g' -i 'Makefile'
...
make -j1 GIT_VERSION="${pkgver}"
````

closed 2018-05-14 by mvp