mvp/uhubctl #100: build fails when source is not downloaded via git

g0hl1n, 2018-05-22:
Since commit 63122bacb531f01c4c3589b36417580fa09428bd building uhubctl fails when its source is not "git cloned" but for example downloaded as an archive from https://github.com/mvp/uhubctl/archive/v2.0.0.tar.gz.

```
fatal: Not a git repository (or any of the parent directories): .git
make: 'uhubctl' is up to date.
```

This is caused by the mandatory git describe in the Makefile.
Therefore IMHO you should change back to a hard-coded version definition or at least provide a fallback.

mvp, 2018-05-24:
I understand where you are coming from. Until recently, version was stored locally. Unfortunately, it was very inconvenient to support and make releases for downstream projects, so I switched to git tag.

I'm open to ideas how to make uhubctl to build properly, yet for version to exist only as git tag (or git tag to be authoritative source).

g0hl1n, 2018-05-28:
Thanks for your feedback.
So if I get it correctly maintaining any type of hard-coded version information in a file is a no-go?
Would some type of "make-release-script" which creates a git tag and changes the version information inside a "fall-back file" be acceptable?
Something like the following (NOT TESTED) script

    #!/bin/bash
    if [ -z "$1" ]; then
        echo "No version given!"
        exit
    fi
    v=$1
    git tag -l | grep "^v${v}$" >/dev/null
    ret=$?
    if [ $ret -eq 0 ]; then
        echo "Version already exists!"
        exit
    fi
    echo "preparing uhubctl version $v"
    echo "$v" > program_version
    git add program_version
    git commit -s -m "uhubctl $v" program_version
    git tag "v$v"

This file (program_version) should then be used by the Makefile if not git-tag was found...

The problem I have is that without the git information (which normally isn't inside the source archive) there's IMHO no possibility to recover the program version...

pabs3, 2018-11-15:
I recommend using autorevision to solve this issue:

https://autorevision.github.io/

mvp, 2018-11-15:
Perhaps, but sounds like using a hammer to kill a fly... I don't want to depend on another external tool to make it work...

pabs3, 2018-11-15:
BTW, after clicking through the package links on the uhubctl repology
page, I see that all the distro packages are working around this in
various different ways:

https://repology.org/metapackage/uhubctl/packages

-- 
bye,
pabs

https://bonedaddy.net/pabs3/

mvp, 2018-11-15:
Yes, I hear you. I guess easiest solution to fix this is go back - maintain version in a header or makefile, and bump it for next release :(
I will think of better ways to fix it...

mvp, 2019-04-28:
Fixed in e93d1a0.

closed 2019-04-28 by mvp