mvp/uhubctl #101: Anker 3-Port USB 3.0

evilpete, 2018-06-06:
this works 

Model : A7514041
Anker 3-Port USB 3.0 w/ Gigabit Ethernet Port Network Adapter 

https://www.amazon.com/gp/product/B00PC07T02

Current status for hub 20-17 [2109:0812 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 0203 power 5gbps U0 enable connect [0bda:8153 Realtek USB 10/100/1000 LAN 000002000000]
  Port 4: 02a0 power 5gbps Rx.Detect

---

Other info : 

USB3.0 Hub             :

  Product ID:	0x0812
  Vendor ID:	0x2109  (VIA Labs, Inc.)
  Version:	b.e1
  Speed:	Up to 5 Gb/sec
  Manufacturer:	VIA Labs, Inc.         
  Location ID:	0x14600000 / 22
  Current Available (mA):	900
  Current Required (mA):	0
  Extra Operating Current (mA):	0

USB 10/100/1000 LAN:

  Product ID:	0x8153
  Vendor ID:	0x0bda  (Realtek Semiconductor Corp.)
  Version:	30.00
  Serial Number:	000002000000
  Speed:	Up to 5 Gb/sec
  Manufacturer:	Realtek
  Location ID:	0x14630000 / 29
  Current Available (mA):	900
  Current Required (mA):	256
  Extra Operating Current (mA):	0

evilpete, 2018-06-06:
update:  I can control the power to a usb-flash drive (with a indicator light)

but can not control the power  a usb LED reading light

mvp, 2018-06-06:
Perhaps its better to not list such products in first place. uhubctl is about controlling power. What's the point of using USB devices which are not really capable of controlling power? :(

evilpete, 2018-06-06:
i agree

mvp, 2018-06-08:
Closing this, unfortunately. Please do report any other hubs that support PPPS properly, including VBUS off.

closed 2018-06-08 by mvp