mvp/uhubctl #102: TROND D4 4-Port USB Hub 3.0

evilpete, 2018-06-06:
TROND D4 4-Port USB Hub 3.0 

all four ports can be controlled although the individual  USB power indicator lights are not effected even when port is off

Current status for hub 2-1 [0bda:5401 Realtek USB3.0 Hub, USB 2.00, 4 ports]

https://www.amazon.com/TROND-D4-Individual-Indicators-Detachable/product-reviews/B011TT9YB6

LED reading lights are not effected, the power stays on (VBUS stays on )

mvp, 2018-06-06:
That's what I feared. Almost all new USB 3 hubs are reporting to support PPPS in hub descriptor, but only few actually have proper circuitry to control VBUS. I would rather not list such hubs in first place.

mvp, 2018-06-08:
Closing this, unfortunately. Please do report any other hubs that support PPPS properly, including VBUS off.

closed 2018-06-08 by mvp