mvp/uhubctl #103: J5Create hub not working in usb3 mode

matrix200, 2018-06-14:
J5Create hub works for power only if plugged into usb 2.0 mode.
If plugged into usb 3.0 power is only being turned off for the usb 2.0 virtual hub, leaving the power on.
Tried both with duality on and off.
What am I doing wrong?

matrix200, 2018-06-14:
Also on USB 2.0 mode, the device won't stay powered off for long.
After a random period time (usually a few minutes) it turns back on.
Centos 6.9

matrix200, 2018-06-14:
Just tried with r 150. Will update you if it sticks.

mvp, 2018-06-14:
No, master is slightly ahead of 2.0.
Regarding -r N - it simply sends power off many times in a row. Some usb device drivers in Linux kernel don't like when power is turned off and are trying to turn it back on. -r N should suppress that.

matrix200, 2018-06-14:
OK, just built from master, still not working with USB3, will report if power off state sticks with USB2.

mvp, 2018-06-14:
Yes, -r N won't help if kernel would try to turn device back on few minutes later.
Re device driver still showing up after power is off, that's almost normal (not really) - see more discussion about this in this LKML thread: https://www.spinics.net/lists/linux-usb/msg157413.html

matrix200, 2018-06-14:
So far it stays off (fingers crossed).
Will report in a few more hours.
What about the USB 3 not controlling power at all though?
I see it trying to power off both USB3 and USB2 hubs, but only USB2 ends up in "off" state.

Re device driver still showing up, a very depressing read.

matrix200, 2018-06-14:
[root@linuxserver /]# uhubctl -l 2-4 -p 1 -a 0
Current status for hub 2-4 [05e3:0617 GenesysLogic USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 2-4 [05e3:0617 GenesysLogic USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 00a0 5gbps Rx.Detect
Current status for hub 1-4 [05e3:0610 GenesysLogic USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
Sent power off request
New status for hub 1-4 [05e3:0610 GenesysLogic USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0000 off

Current status for hub 2-4 [05e3:0617 GenesysLogic USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 00a0 5gbps Rx.Detect
Sent power on request
New status for hub 2-4 [05e3:0617 GenesysLogic USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
Current status for hub 1-4 [05e3:0610 GenesysLogic USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0000 off
Sent power on request
New status for hub 1-4 [05e3:0610 GenesysLogic USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power

The thing is though, even if the port doesn't show "power" in its status, device connected to it is still powered (tried both the "dumb" power cable that powers my usb docking station and its data cable.

matrix200, 2018-06-14:
Almost forgot to mention:
I tried running the two commands separately.
No joy.
The device still stays powered.

matrix200, 2018-06-14:
Sorry, I meant I ran the commands from terminal one after another (with the -e switch) one for usb3 and one for usb2.
It didn't work.
The thing is, the port doesn't change to off state, but goes into that weird Rx.Detect state, which obviously somehow keeps power on.

mvp, 2018-06-14:
According to USB3 spec, port link state bits (USB_PORT_STAT_LINK_STATE mask) are valid only if port is enabled - in our case it is not.
Most hubs will have port link state as SS.Disabled when it is not enabled, but some may have Rx.Detect.
Perhaps I should change uhubctl to not display port state bits at all if it is not enabled.

That said, we have this weird combination that port is not enabled, but power is still on. This could be non-compliant hardware :(

matrix200, 2018-06-15:
Hmm, why does it work in USB2 mode then?
Well "kind of" works, since there is a problem of random turning back on (which I will continue to investigate).
At any rate, I guess this means J5create goes off the supported devices list?

mvp, 2018-06-15:
Well, since it does work in USB2 mode, it should stay in the list. I will add a note about it when I have time on this weekend.

mvp, 2018-06-15:
Added comment about USB2 only in 1709413.

closed 2018-06-15 by mvp

parkerlreed, 2018-10-02:
Can I get some clarification on this? 

I have this hub plugged into a 3.0 port and can toggle a 2.0 device. Does this mean I can't toggle a 3.0 device?

mvp, 2018-10-02:
Each USB 3 hub is detected as 2 virtual devices: one is USB 3, another USB 2. They share the same physical ports. To control power on such hubs, we must switch power off on both USB 2 and USB 3 hubs.

It is possible that it actually works, and previous report wasn't correct. Best way to confirm is to plug USB light or phone and check if light goes off or phone stops charging when port is turned off.

matrix200, 2018-10-02:
Thats interesting.
I am the guy who reported that it doesnt work for me.
Could you please provide the output of lsusb with your hub included?
Also what OS are you running?
Mine is Centos 6.9 and it worked only if plugged into usb 2 port.

parkerlreed, 2018-10-02:
```
[parker@yoga710 ~]$ uname -a
Linux yoga710 4.19.0-rc5-g9ffc59d57228 #1 SMP PREEMPT Thu Sep 27 11:41:48 EDT 2018 x86_64 GNU/Linux
[parker@yoga710 ~]$ head -1 /etc/os-release 
NAME="Arch Linux"
[parker@yoga710 ~]$ lsusb
Bus 002 Device 005: ID 0b95:1790 ASIX Electronics Corp. AX88179 Gigabit Ethernet
Bus 002 Device 004: ID 05e3:0617 Genesys Logic, Inc. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 138a:0094 Validity Sensors, Inc. 
Bus 001 Device 003: ID 8087:07dc Intel Corp. 
Bus 001 Device 022: ID 17ef:608c Lenovo 
Bus 001 Device 016: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 008: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
[parker@yoga710 ~]$ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 5000M
    |__ Port 3: Dev 4, If 0, Class=Hub, Driver=hub/4p, 5000M
        |__ Port 4: Dev 5, If 0, Class=Vendor Specific Class, Driver=ax88179_178a, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/12p, 480M
    |__ Port 2: Dev 8, If 2, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 2: Dev 8, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 2: Dev 8, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 3: Dev 16, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 2: Dev 22, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 2: Dev 22, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 7: Dev 3, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 7: Dev 3, If 1, Class=Wireless, Driver=btusb, 12M
    |__ Port 8: Dev 5, If 0, Class=Vendor Specific Class, Driver=, 12M
```

matrix200, 2018-10-02:
Hmm, on its face the same id's as with mine.
I am completely at a loss why it wont work for me.
Either it is my older kernel I compiled against  which is of 2.6.* variety, or my specimen has a different hardware despite having the same id.

mvp, 2018-10-02:
@matrix200, Linux kernel 2.6 sounds ancient.
I believe you need something more modern for USB 3 to work well, e.g. kernel 3.x or 4.x.

matrix200, 2018-10-03:
In my defense, that kernel is supposed to be more modern then its version number applies as Redhat devs are backporting many changes from more modern kernels into it.
Anyway I now got around to ordering Amazon basic 4 port hub which is also supposed to work and will test it once it arrives.