mvp/uhubctl #105: Plugable 7-port USB hub does not work https://plugable.com/products/usb3-hub7c/

jensolsson, 2018-06-14:
Just bought 4 7-port USB hubs for plugable and they do not work.

Everything looks good but ports are not actually turning on and off, see below.

I also saw that 2109:2813 is not the same as in the supported list which is 2109:0813
Any chance there is a code change needed or should I return these to Amazon ? :)

$ ./uhubctl -l 2-1.2
Current status for hub 2-1.2 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0507 power highspeed suspend enable connect [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 2: 0100 power
  Port 3: 0000 off
  Port 4: 0100 power
Current status for hub 2-1.2 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0507 power highspeed suspend enable connect [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 2: 0100 power
  Port 3: 0000 off
  Port 4: 0100 power

mvp, 2018-06-14:
Your hub is somehow listed twice with the same id? This suggests that you're using old version of libusb. On Mac, make sure to link to libusb 1.0.22. Also, build from master, and try using usb3 mode.

jensolsson, 2018-06-14:
In this case running 2.0.0-7-g448e on ubuntu. Is that to old?

mvp, 2018-06-14:
Try building from master. It should be showing version 2.0.0-13-g51ff

jensolsson, 2018-06-14:
OK done so now, will try the hub again tomorrow

jensolsson, 2018-06-15:
Unfortunately I have tested now and it does not work. I have 3 AmazonBasics hubs connected to the same computer and it works good.

Is there something tweakable in uhubctl or are these hubs useless?

#./uhubctl --version
v2.0.0-13-g51ff
#./uhubctl -l  2-1.2
Current status for hub 2-1.2 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0507 power highspeed suspend enable connect [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 2: 0100 power
  Port 3: 0503 power highspeed enable connect [04e8:6860 SAMSUNG SAMSUNG_Android 4200340c568fa200]
  Port 4: 0100 power
#./uhubctl -l  2-1.2 -p 3 -a 0
Current status for hub 2-1.2 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 3: 0503 power highspeed enable connect [04e8:6860 SAMSUNG SAMSUNG_Android 4200340c568fa200]
Sent power off request
New status for hub 2-1.2 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 3: 0000 off
#./uhubctl -l  2-1.2 -p 3 -a 0
Current status for hub 2-1.2 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 3: 0000 off
Sent power off request
New status for hub 2-1.2 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 3: 0000 off
#./uhubctl -l  2-1.2 -p 3 -a 1
Current status for hub 2-1.2 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 3: 0000 off
Sent power on request
New status for hub 2-1.2 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 3: 0101 power connect [04e8:6860]
#./uhubctl -l  2-1.2 -p 3 -a 1
Current status for hub 2-1.2 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 3: 0503 power highspeed enable connect [04e8:6860 SAMSUNG SAMSUNG_Android 4200340c568fa200]
Sent power on request
New status for hub 2-1.2 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 3: 0503 power highspeed enable connect [04e8:6860 SAMSUNG SAMSUNG_Android 4200340c568fa200]
#

jensolsson, 2018-06-15:
Looking at the sticker on the back on the device, model is :  USB3-HUB7C2C, even though I bought  USB3-HUB7C on amazon... maybe there is a difference

mvp, 2018-06-15:
Ok, I will remove this hub from supported list then. I'm glad at least AmazonBasics is working for you.

mvp, 2018-06-15:
It seems that port is logically turning off. How did you test that it doesn't work for you? Did you measure VBUS? Or checked if phone connected to port is still charging?
Also, did you try to connect it in USB3 mode? You will need actual USB3 upstream port and compliant cable.

jensolsson, 2018-06-15:
Maybe not remove it but comment that the currently available version is not supported?
I rroed with a phone which did not stop charging. 
Also i just used normal switches no specific usb3 ones. Would that be required? It is not required with the amazonbasics hubs i tried

mvp, 2018-06-15:
If you have computer with true USB3 root hub, and use compliant USB3 cable (typically with blue end) to connect your hub, it should be detected as 2 hubs: one USB2, another USB3, something like this:

```
$ uhubctl
Current status for hub 4-4 [0451:8140, USB 3.00, 4 ports]
  Port 1: 06a0 power Rx.Detect
  Port 2: 06a0 power Rx.Detect
  Port 3: 06a0 power Rx.Detect
  Port 4: 06a0 power Rx.Detect
Current status for hub 3-13 [0451:8142, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
```
Both of these hubs are actually 1 physical hub. More details are explained in USB3 duality note in readme: https://github.com/mvp/uhubctl/blob/master/README.md#usb-30-duality-note

If you don't have USB3 upstream port, or use non USB3 compliant cable, you will only see 1 hub, USB2 (what you seem to have now).
Some hubs work better in USB3 mode, while some hubs only work in USB2 mode.

jensolsson, 2018-06-15:
Now when you mention it of course you are right, I always thought the computer had usb3 ports since they are blue, but apparently the root hub is actually usb2

```
# lsusb
Bus 002 Device 004: ID 2109:2811 VIA Labs, Inc. Hub
Bus 002 Device 003: ID 2109:2811 VIA Labs, Inc. Hub
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 024: ID 2109:2811 VIA Labs, Inc. Hub
Bus 001 Device 023: ID 2109:2811 VIA Labs, Inc. Hub
Bus 001 Device 009: ID 2109:2811 VIA Labs, Inc. Hub
Bus 001 Device 003: ID 2109:2811 VIA Labs, Inc. Hub
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

jensolsson, 2018-06-18:
Was going to try the same hub on my mac, but was not able to properly reference the hub... As you can see -l 20-1 references -l 20-16.1   Guess this is a bug? Is there some kind of string search involved?

$ ./uhubctl --version
v2.0.0-15-g1709

$ ./uhubctl -l 20-1
Current status for hub 20-16.1 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 20-1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0503 power highspeed enable connect [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power

mvp, 2018-06-18:
By default, uhubctl tries to operate on both hub you referenced and its USB2/USB3 dual (this only applies to USB3 physical hubs).
On your machine, it decided that hub 20-16.1 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports] is actually dual to 20-1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports], and that's why it is showing both.
You can disable this duality using option -e. But remember that if you want to turn power off, you will have to execute uhubctl twice, something like this:

    uhubctl -e -l 20-1 -p 4 -a 0
    uhubctl -e -l 20-16.1 -p 4 -a 0

That said, dual hub detection is not perfect. It is possible that it may match incorrect hub - you may have hit a bug here.
On Linux, it is relatively reliable: hub and its dual will almost always have location that differs only in bus number, e.g. 3-2.3.4 and 4-2.3.4. On Mac, it is very difficult to predict...
At any rate, using -e for hub and its true dual is most reliable, but awkward way to control it.

jensolsson, 2018-06-18:
OK so that sounds understandable, but if I do this I get all kinds of hubs listed. Should'nt I only get 20-1.1 ?

$ ./uhubctl -l 20-1.1
Current status for hub 20-1.1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0103 power enable connect [0b0c:003f Todos Data System AB Handelsbanken card reader]
Current status for hub 20-16.1 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 20-1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0503 power highspeed enable connect [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 20-16 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0203 power 5gbps U0 enable connect [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 20-6.1 [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0103 power enable connect [1050:0407 Yubico Yubikey 4 OTP+U2F+CCID]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0103 power enable connect [067b:2303 Prolific Technology Inc. USB-Serial Controller]
Current status for hub 20-21.4 [2109:0812 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 20-21.1 [2109:0812 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 20-6.4 [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0103 power enable connect [05ac:1392 Apple Inc. Apple Watch Magnetic Charging Cable XXXXX]

jensolsson, 2018-06-18:
I had some success now using the -e switch. 

Using USB2 devices I can iluminate end deluminate the led on the device. The connected perihperal also seem to disconect but I dont have the usb multimeter available so need to try this again tomorrow.

Using a USB3 device, the device will auto connect again after sending /uhubctl -e -l 20-16.1 -p 4 -a 0
Strange...

mvp, 2018-06-19:
Good news is that hub hardware is actually working as intended in USB3 mode.
For USB2 only mode, if you specify explicit location like `uhubctl -l 20-1.1 -p 4 -a 0`, error message `changing port state for multiple hubs at once is not supported` should not be possible - it must be a bug.
Can you please show full `uhubctl` output for Linux box? Or, at least output for `uhubctl -l 20-1.1`. Thanks!

jensolsson, 2018-06-19:
Just to be clear the output above is for the osx box. Understand that my message was a bit unclear.

On the linux box I can try again soon I hope

mvp, 2018-07-06:
Can you please provide more details about possibly duplicated hubs listed?
My biggest concern is fixing error message `changing port state for multiple hubs at once is not supported` when you specify hub location.

mvp, 2018-07-29:
Closing by no activity from submitter. Please feel free to reopen if you manage to reproduce duplicated hubs listed issue.

closed 2018-07-29 by mvp