mvp/uhubctl #108: power off the usb cause system crash

mcdull, 2018-06-19:
I have built the uhubctl on a raspberry zero wh and would like to control a 7 port orico.  I installed the latest libusb and build from master.

When issuing the command, some will response instantly, some may delay for 10 seconds, and sometime it will just crash.  The image I use is raspbx, and should be quite similar to the original raspbain.   
`
root@raspbx:~/uhubctl# apt-get upgrade libusb-1.0-0-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
libusb-1.0-0-dev is already the newest version (2:1.0.21-1).
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@raspbx:~/uhubctl# ./uhubctl  --version
v2.0.0-15-g1709
root@raspbx:~/uhubctl# ./uhubctl
Current status for hub 1-1.1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0503 power highspeed enable connect [12d1:1001 HUAWEI Technology HUAWEI Mobile]
  Port 2: 0503 power highspeed enable connect [12d1:1001 HUAWEI Technology HUAWEI Mobile]
  Port 3: 0503 power highspeed enable connect [12d1:1001 HUAWEI Technology HUAWEI Mobile]
  Port 4: 0503 power highspeed enable connect [12d1:1001 HUAWEI Technology HUAWEI Mobile]
Current status for hub 1-1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0503 power highspeed enable connect [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [12d1:1001 HUAWEI Technology HUAWEI Mobile]
root@raspbx:~/uhubctl# ./uhubctl  -a off -p 1 -l 1-1.1
Current status for hub 1-1.1 [2109:2813, USB 2.10, 4 ports]
  Port 1: 0503 power highspeed enable connect [12d1:1001 HUAWEI Technology HUAWEI Mobile]
Sent power off request
New status for hub 1-1.1 [2109:2813, USB 2.10, 4 ports]
  Port 1: 0000 off
root@raspbx:~/uhubctl# ./uhubctl  -a off -p 2 -l 1-1.1
Current status for hub 1-1.1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 2: 0503 power highspeed enable connect [12d1:1001 HUAWEI Technology HUAWEI Mobile]
Sent power off request
New status for hub 1-1.1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 2: 0000 off
root@raspbx:~/uhubctl# ./uhubctl  -a off -p 3 -l 1-1.1
Current status for hub 1-1.1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 3: 0503 power highspeed enable connect [12d1:1001 HUAWEI Technology HUAWEI Mobile]
Sent power off request
New status for hub 1-1.1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 3: 0000 off
root@raspbx:~/uhubctl# ./uhubctl  -a off -p 4 -l 1-1.1
Current status for hub 1-1.1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 4: 0503 power highspeed enable connect [12d1:1001 HUAWEI Technology HUAWEI Mobile]

`

mvp, 2018-06-19:
Are you trying to control internal hub or external one? Raspberry Pi Zero internal hub doesn't support proper power switching. As for external hubs, I'm not aware of Orico. Any hub from current supported devices should work.

SlySven, 2018-06-20:
> Are you trying to control internal hub or external one?

As far as I recall the RPi **Zeros** are like the **Model 'A'** variants of their bigger brothers in that **they do not *have* an internal hub** - only a single USB port; the **Model 'B'** variants of the full-size RPs of course combine a 4 (2 on older models) USB 2.0 Hub which also contains the wired Ether-networking port so that also explains why the 'A's lack built-in wired network capabilities and why the Zero doesn't either.

mvp, 2018-07-06:
Sorry, but I have not seen any response about details in more than 2 weeks.
Feel free to reopen if you can provide more details about your crashes.

closed 2018-07-06 by mvp