mvp/uhubctl #109: Port is powered off, and turns back on automatically

McLion-Himself, 2018-07-04:
Hi
I'm trying to turn selective ports off on a LAN9514 controlled from RPi3-CM using Raspbian Stretch.
The power of a selective port goes low but automatically turns back on.
Command used:` sudo ./uhubctl -a off -p 4`
What can I do to keep the port off?

I actually would like it to be inactive with power on to still use it for charging, but this seems only to be possible (bind/unbind) once a device is connected and I need to turn it off before a device is connected.

Thanks
McL

McLion-Himself, 2018-07-04:
Never mind ... I just found this: https://github.com/mvp/uhubctl/issues/96
It works to turn it off with higher repeat values than about 200.

However, I leave this open in case someone has a solution to only disable the communication link while no device is attached and leave power on to use it fro charging but not enumerate any device.

mvp, 2018-07-04:
There is no way to disable just data link. However, some hubs (which don't conform to USB spec) are doing just that when they report that power is off.

closed 2018-07-04 by mvp

Cyoshi, 2019-11-19:
i have a like problem even the -r 500 is not working

mvp, 2019-11-19:
Note that some people reported needing 1000 and higher