mvp/uhubctl #110: Mac Mini, PPPS

serjepatoff, 2018-08-13:
Not an issue, just a question. Does anybody know what models of Mac Mini does support PPPS on the internal 4-port hub (if any)? 

(Yes, I see that hardware compatibility list doesn't include Mac Mini, but it also misses Macbook Pro 2015 which is definitely supported).

mvp, 2018-08-13:
@serjepatoff, I don't know how you found that Macbook Pro supports PPPS - I have access to Macbook Pro late 2013 and Macbook Pro 2017 models, they both do not support PPPS. Note that white Apple keyboard does support PPPS, and it is often used with these laptops - you may have mistaken this keyboard as onboard device, but it isn't.

If you have different experience, please do report and I would be happy to add it (Mac Mini, Macbook Pro or any other device) to compatibility list. Please provide as much detail as possible, including exact model numbers, output from uhubctl and whether turning power off stops phone charging or turns off USB light. Thanks!

serjepatoff, 2018-08-13:
I don't have MBP now, but as far as I can remember (*) uhubctl worked perfectly 1.5 years ago on MBP/Sierra. 

*: Yes, it can be a false memory, and I can actually remember the keyboard or ext. hub in Apple 27'' Display connected to MBP.

mvp, 2018-08-13:
Yes, Apple Thunderbolt 27" display supports PPPS and is included in compatibility list. Apple white keyboard as well. But MacBook internal USB hub doesn't support PPPS.

serjepatoff, 2018-08-13:
Also do you know if Plugable USB3-HUB7BC actually supports PPPS? Or just reports it as supported, but lacks per-port MOSFET switches?

mvp, 2018-08-13:
I don't know for sure. Originally, we had some issues with USB 3, which now should be resolved. USB 3 hubs require turning off both USB 3 and USB 2 virtual hubs, which we didn't do originally. Current version should do it properly. It should work if hardware is actually there.

mvp, 2018-08-13:
Closing this as not actionable. Please do report new devices (e.g. Mackbook Pro or Mac Mini) if they seem to work with uhubctl. Thanks!

closed 2018-08-13 by mvp

serjepatoff, 2018-08-14:
Thank you for help!