mvp/uhubctl #111: long time to enable port if there is no internet-connection

nfstoney, 2018-08-14:
Hello
My system is Raspberry PI CML with USB-Hub LAN9514  with Raspbian 9.3 stretch, uhubctl v2.0.0-15-g1709.
Port4 will disabled by boot up with command: sudo /home/pi/norfolk/uhubctl/uhubctl -a off -p 4 -r 500

If I want to enable the port 4 with command: sudo /home/pi/norfolk/uhubctl/uhubctl -a on -p 4
there is a long time until the uhubctl gives a feedback with port overview.

Overview:
Current status for hub 1-1 [0424:9514, USB 2.00, 5 ports]
  Port 4: 0100 power
Sent power on request
New status for hub 1-1 [0424:9514, USB 2.00, 5 ports]
  Port 4: 0100 power

This behavior is only given if there is no internet-connection.
If the internet-connection is running the feedback with port-overview will show immediately.

Port1 of this USB-Hub is Ethernet-port.

Can you reproduce this behavior or is there a simple explanation?

Best regards
nfstoney

mvp, 2018-08-14:
First off, all Raspberry Pi models (except for 3B+) should use port 2 to control power for all USB ports (port 1 for Ethernet+WiFi power). This doesn't apply to external hubs supporting PPPS, but they should use explicit hub location with -l option.

Now as to why it is slow, I don't know, this may need some tracing where slowdown takes place. It was noted by other Raspberry Pi users that enabling power may work better if you add -R option, which should completely reset USB bus after power on.

mvp, 2018-08-23:
Closing for inactivity from reporter. Please feel free to reopen if you think issue is not resolved and you can provide more info about this failure.

closed 2018-08-23 by mvp