mvp/uhubctl #113: udev keeps devices after port power down

joeyh, 2018-09-09:
I noticed that after powering down a port with a disc drive attached, udev does not get any event, so it keeps the device files around. However, trying to access the device files will lead to an IO error. This presents a problem when powering the port back up; normally the device files will only appear once the drive has spun up, but since they were never cleaned up, you can't know when it's safe to use the drive again.

mvp, 2018-09-09:
This is Linux kernel issue. It may be eventually fixed in kernel, see more discussion here: https://www.spinics.net/lists/linux-usb/msg157413.html
Long story short: kernel USB driver knows about power off, but doesn't send notification about it to udev.

mvp, 2018-09-09:
I had considered adding Linux- specific code to uhubctl to do something like write 1 to `/sys/bus/usb/devices/1-4.1/remove`, as discussed in post by Harald Welte above. After numerous experiments I wasn't able to find a way to power on such device again with uhubctl without physical replugging.

Note that other OS like Mac or FreeBSD do not have this problem.

joeyh, 2018-09-19:
I agree this is not really a bug in uhubctl, anyway I found a
workaround. Running eg "udevadm trigger --action=remove /dev/disk/by-label/passport"
after turning off the hub port makes udev remove the link to the device
file (though it keeps the device file still).

When the port's powered back up, it then adds back the link, after the
drive is spun up. So anything relying on that link to mount the drive
will only mount it when it's really available.

It might be possible for uhubctl to trigger these removals itself,
and so avoid the problem at least for USB drives if not other devices.

-- 
see shy jo

matrix200, 2018-10-10:
@joeyh, I have the same problem now (Centos 6.9).
Just to clarify, does triggering the event prevent these io errors from appearing in dmesg?
Also as a general question, are these errors somehow indicative of a possible damage to the device?
File corruption, mechanical damage to the disk etc?
Or perhaps this can be ignored since the filesystem is unmounted beforehand anyway?