mvp/uhubctl #115: Raspberry Pi Issue with USB 3.0 Hubs

SlyLaM, 2018-09-18:
Hi,
it should be great to add a column for checking Raspberry pi compatibility for Hubs.
I just bought an AmazonBasics | HU3770V1 and it's not working at all. Even if it's USB 2.0 backward compatible, it's not recognized by Rasp Pi 3B+ (which as no Usb3 on it).

Thanks.

mvp, 2018-09-18:
I see no reason why it would not work with RPi3. Are you sure your cables are good?
Also, you need to make sure you provide external power to AmazonBasics hub, otherwise it won't work.
Some hubs can get by with just using power provided by RPi, but USB3 hubs generally need more power and can't function normally without having its own power source.

SlyLaM, 2018-09-18:
I have a new Pi3 B+ and the hub won't work, and of course there's an external power.
This has been reported by several users on amazon discussions, and of course I didn't saw this before ...
Go to:
https://www.amazon.com/AmazonBasics-Port-USB-Power-Adapter/dp/B00E6GX4BG
and search "Raspberry" in the comments or questions.

mvp, 2018-09-19:
Hmm - this is really weird. Not sure who to blame here - RPi, AmazonBasics or simply Linux kernel USB driver.
AmazonBasics does work with other USB2 only Linux hosts, so this should be RPi issue.

Going by your logic, we need to provide full cross matrix of what hub works with which host product - certainly not possible to maintain within uhubctl scope.

Perhaps simpler solution is to link to https://elinux.org/RPi_Powered_USB_Hubs in uhubctl device table for RPi?

mvp, 2018-09-19:
Actually, if we intersect uhubctl supported devices table and problematic PRi hubs table https://elinux.org/RPi_Powered_USB_Hubs#Problem_USB_Hubs, there is only one hub that is supported by uhubctl and is not supported by RPi - AmazonBasics (however one listed there is 4 port, we list 7 port one).

It means simplest solution is to put a note against AmazonBasics hub in our table that it may not work with RPi - I will do that.

mvp, 2018-09-19:
Added a note to supported devices table in dda0ab22f248a8df50d.

closed 2018-09-19 by mvp