mvp/uhubctl #117: uhubctl requires sudo for the first run

ptsneves, 2018-09-30:
I was having some difficulties pinning down a problem where when I connected my Amazon Basics 7 port-hub the ports switching was not working correctly with uhubctl. Instead of giving an error, uhubctl, just finishes and reports that the port is off when in fact it is not.

It came down to using sudo or not. I checked that the reporting is the same with sudo and without with pexpect just to confirm.

I am using the latest master a9474fe12fe5f951c55dc10fd5cf5b770ba61a1d. I am also using a udev rule that half works, because with the udev rule i do not see the error of uhubctl complaining about permissions. Even so, the uhubctl incorrectly reports success.

The udev is very simple
SUBSYSTEMS=="usb", ATTR{idVendor}=="2109", ATTR{idProduct}=="2811", MODE="0666"

I guess there must be some error code which is not being correctly checked but I do not have the knowledge to investigate this. If you give me a tip I can investigate it myself.

Also consider changing the help message to be explicit that the idVendor and idProduct can vary with whichever hub the user is using, and he should correct it for his case.

Thanks a lot for the work

mvp, 2018-09-30:
Note that USB 3 hubs are detected as 2 virtual hubs, one USB 2 and another is USB 3. uhubctl needs to control power on both for power changes to take effect. However, these hubs may have different idProduct. You should either add 2 lines for each idProduct, or remove ATTR{idProduct} restriction from your rule for this to work.

ptsneves, 2018-09-30:
Ouch your are right!
Thanks a lot for the hint and fast answer

closed 2018-09-30 by ptsneves