mvp/uhubctl #119: doesn't seem to work with rpi 3b

dotps1, 2018-10-22:
I built and compiled, and I "think" im using correctly:

list devices, the port I want off is not plugged in.
```
pi@octopi:~/uhubctl $ sudo ./uhubctl
Current status for hub 1-1 [0424:9514, USB 2.00, 5 ports]
  Port 1: 0503 power highspeed enable connect [0424:ec00]
  Port 2: 0503 power highspeed enable connect [1e4e:0102 Etron Technology, Inc. USB2.0 Camera]
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0100 power
```

seems legit, I plug in my 3d printer (I don't want the rpi powering the controller board at all).

```
pi@octopi:~/uhubctl $ sudo ./uhubctl
Current status for hub 1-1 [0424:9514, USB 2.00, 5 ports]
  Port 1: 0503 power highspeed enable connect [0424:ec00]
  Port 2: 0507 power highspeed suspend enable connect [1e4e:0102 Etron Technology, Inc. USB2.0 Camera]
  Port 3: 0100 power
  Port 4: 0103 power enable connect [0403:6001 FTDI FT232R USB UART AH06OEVF]
  Port 5: 0100 power
```

so it looks like port 4 is the one I want to disable:

```
pi@octopi:~/uhubctl $ sudo ./uhubctl -a off -p 4
Current status for hub 1-1 [0424:9514, USB 2.00, 5 ports]
  Port 4: 0103 power enable connect [0403:6001 FTDI FT232R USB UART AH06OEVF]
Sent power off request
New status for hub 1-1 [0424:9514, USB 2.00, 5 ports]
  Port 4: 0000 off
```

the device still has power, am I missing something?

thanks.

dotps1, 2018-10-22:
if I turn off port 2, it turns off all the usb ports, is that the only way I can do it?

mvp, 2018-10-22:
This is duplicate for https://github.com/mvp/uhubctl/issues/64.
This is quirk of Raspberry PI internal hardware design. USB port 1 is wired to control power on network (wifi+Ethernet), and USB port 2 is wired to control power on all remaining USB ports - 3,4,5.
Trying to control ports 3,4,5 won't do anything, unfortunately.

mvp, 2018-10-22:
Note that you can always connect any hub from supported devices list to RPi, and then control power for your devices separately on that external hub. The only trouble is that almost any external hub will be probably bigger than RPi and will cost about the same as RPi.

closed 2018-10-22 by mvp

dotps1, 2018-10-22:
ah.  Thanks for letting me know, that's a bummer.  I was hoping to only shut down one port and I have a web cam with leds on it as well.

thanks.

cbrherms, 2018-11-05:
Well, that's a bummer =/