mvp/uhubctl #120: Seeing coredump on Fedora 28 on RPi

xxedgexx, 2018-10-27:
I'm seeing uhubctl dump core:

Oct 26 22:04:01 pidev.subtraverse.intra systemd-coredump[20865]: Process 20862 (uhubctl) of user 65534 dumped core.

                                                                 Stack trace of thread 20862:
                                                                 #0  0x0000ffffa42dc014 strjoin_real (libudev.so.1)
                                                                 #1  0x0000ffffa42d56e4 chase_symlinks (libudev.so.1)
                                                                 #2  0x0000ffffa42ce704 device_set_syspath (libudev.so.1)
                                                                 #3  0x0000ffffa42ceb00 sd_device_new_from_syspath (libudev.so.1)
                                                                 #4  0x0000ffffa42cc4bc enumerator_scan_dir_and_add_devices (libudev.so.1)
                                                                 #5  0x0000ffffa42cc8ac enumerator_scan_dir (libudev.so.1)
                                                                 #6  0x0000ffffa42ccb5c device_enumerator_scan_devices (libudev.so.1)
                                                                 #7  0x0000ffffa448ab18 linux_udev_scan_devices (libusb-1.0.so.0)
                                                                 #8  0x0000ffffa4489280 op_init (libusb-1.0.so.0)
                                                                 #9  0x0000ffffa44805d0 libusb_init (libusb-1.0.so.0)
                                                                 #10 0x0000000000402b04 main (uhubctl)
                                                                 #11 0x0000ffffa43249a4 __libc_start_main (libc.so.6)
                                                                 #12 0x000000000040128c $x (uhubctl)
                                                                 #13 0x000000000040128c $x (uhubctl)
[](url)

mvp, 2018-10-27:
`libusb_init` is call to initialize libusb, and happens really early. It seems there is something wrong with libudev on this Fedora.

xxedgexx, 2018-10-27:
This is the version:

libusb-0.1.5
systemd-udev-238-9.git0e0aa59

mvp, 2018-10-27:
libusb 0.1.5 doesn't seem right. uhubctl requires libusb 1.0.12 or higher.

mvp, 2018-10-27:
In that case, my call is on bad libudev. I believe that even 3 line C program only calling `libusb_init()` should crash the same.

mvp, 2018-10-30:
@xxedgexx , can you please try compiling this and running following C program:

    #include <libusb-1.0/libusb.h>

    int main(int argc, char *argv[]) {
        rc = libusb_init(NULL);
        libusb_exit(NULL);
        return 0;
    }

I expect that it should crash on your platform. If it does, this is not an issue with uhubctl.

mvp, 2018-10-31:
@xxedgexx, I am closing this for lack of response.
Please feel free to reopen if you believe this is real issue that can be solved by changing uhubctl.

closed 2018-10-31 by mvp

xxedgexx, 2018-11-06:
Just a follow up on this.  Fedora released an update to systemd-udev.  Things seem better now.  Here's the changeling:

* Mon Oct 29 2018 Zbigniew Jędrzejewski-Szmek <zbyszek@in.waw.pl> - 238-10.git438ac26
- Fix a local vulnerability from a race condition in chown-recursive (CVE-2018-15687, #1643367)
- Fix a local vulnerability from invalid handling of long lines in state deserialization (CVE-2018-15686, #1643372)
- Fix a remote vulnerability in DHCPv6 in systemd-networkd (CVE-2018-15688, #1643362)
- Downgrade logging of various messages and add loging in other places
- Many many fixes in error handling and minor memory leaks and such
- Fix typos and omissions in documentation
- Various smaller improvements to unit ordering and dependencies
- Handling of invalid (intentionally corrupt) dbus messages is improved, fixing potential local DOS avenues
- The target of symlinks links in .wants/ and .requires/ is now ignored. This fixes an issue where
  the unit file would sometimes be loaded from such a symlink, leading to non-deterministic unit contents.
- Filtering of kernel threads is improved. This fixes an issues with newer kernels where hybrid kernel/user
  threads are used by bpfilter.
- Catalog entries for the journal are improved (#1639482)