mvp/uhubctl #121: Two new suppoerted USB-Hubs

yrjola, 2018-10-29:
I bought today two externally powered hubs from my local computer store in order to test if any of them would work and to my surprise both support uhubctl.
TP-Link USB 3.0 7-Port Hub, Model UH700 and Transcend 4-Port Hub Model TS-HUB3K.
Outputs from uhubctl and lsusb in the attachment.
[USB_PPPS.pdf](https://github.com/mvp/uhubctl/files/2526641/USB_PPPS.pdf)

mvp, 2018-10-29:
Thanks for the report, @yrjola! Can you please try connecting e.g. your phone or USB light to the hub port and turning it off with uhubctl? I'd like to know if these hubs actually cut off VBUS to port - check if phone stops charging or USB light comes off.

Also, for Transcend TS-HUB3K, can you please connect it to USB3 upstream port using USB3 compliant cable? It seems you have connected it only using USB2. If you used true USB3 connection, it may mean that USB3 part does not support PPPS.

Thanks!

yrjola, 2018-10-30:
Yep,

I did have a Lacie SSD-drive connected to both hubs and it did die out.
That due to reports I read which indicated that the manufacturer might have
saved by leaving the actual power circuitry out...

What comes to USB 2/3, I did have an USB3 cable, but cubietruck, which is
my target HW for my home NASish device with a SATA-drive and multiple
USB-drives which I can turn off to save the disks/power, does have only
USB2, oh well.

I'll make the USB3 test asap, since it is valuable information to my office
usage also.

isto

mvp, 2018-10-30:
Note that SSD drive dying out may not be 100% reliable test. Most USB devices that uhubctl will detect will cut off data connection (which will also manifest in SSD drive "dying out"). However, it may not necessarily mean that power is completely cut off (VBUS is off).
Good tests for VBUS off are:
1. Use voltmeter and check that voltage is 0 on USB output pins
2. Connect USB light and check that it stops shining or USB fan stops spinning
3. Connect phone and check that it stops charging when USB port is turned off

yrjola, 2018-10-31:
Hello,

I finally got the TS-HUB3K tested on a computer having USB3 and USB2 ports,
this time using an ESD CANbus dongle.
Unfortunately You're right, power control seems to work only with USB2-port.
See attachment.

isto

mvp, 2018-10-31:
@yrjola , I don't see any attachment. Actually, looking deeper, you didn't connect TP-Link hub to USB3 either.
All USB3 hubs I've seen so far can work in 2 modes: USB2 only, where they are detected as USB 2.1 devices - this happens if you connect to USB2 (non-USB3) upstream port, or if you use non-compliant USB3 cable.
When you connect USB3 hub to USB3 upstream using compliant cable, host should detect 2 devices on a bus: USB 2.1 hub, and USB 3.0 hub. These hubs appear to be independent, but they share the same physical ports. For example, this is output from one of my hubs:

```
Current status for hub 4-5 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0203 power 5gbps U0 enable connect [0951:1666 Kingston DataTraveler 3.0]
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
```
This is single device, but is detected as 2 virtual hubs.

yrjola, 2018-11-01:
Ok, need to check what happened with the Pdf.

yrjola, 2018-11-01:
Fu.k the attachments,it was included in the post I sent, I wonder what
happened.
Anyway, ascii with courier is harcore wysiwyg<g>
Here You go:

>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
Hub in USB2 (black connector in thinkpad)

isto@isto-ThinkPad-W510:~/uhubctl/uhubctl-master$ sudo ./uhubctl
Current status for hub 2-1.2 [8564:4100 Transcend USB2.1 Hub, USB 2.10, 4
ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [0ab4:0010 esd electronic
system design gmbh (www.esd-electronics.com) USB2292 HJ005337]
Current status for hub 2-1 [8087:0020, USB 2.00, 8 ports]
  Port 1: 0100 power
  Port 2: 0503 power highspeed enable connect [8564:4100 Transcend USB2.1
Hub, USB 2.10, 4 ports]
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0103 power enable connect [17ef:1003 Lenovo Integrated Smart Card
Reader]
  Port 6: 0100 power
  Port 7: 0100 power
  Port 8: 0100 power
Current status for hub 4 [1d6b:0003 Linux 4.15.0-20-generic xhci-hcd xHCI
Host Controller 0000:0f:00.0, USB 3.00, 2 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
Current status for hub 3 [1d6b:0002 Linux 4.15.0-20-generic xhci-hcd xHCI
Host Controller 0000:0f:00.0, USB 2.00, 2 ports]
  Port 1: 0100 power
  Port 2: 0100 power
Current status for hub 1-1 [8087:0020, USB 2.00, 6 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0103 power enable connect [147e:2016 UPEK Biometric Coprocessor]
  Port 4: 0103 power enable connect [0a5c:217f Broadcom Corp Broadcom
Bluetooth Device 78DD08B2996C]
  Port 5: 0100 power
  Port 6: 0507 power highspeed suspend enable connect [17ef:480f Chicony
Electronics Co., Ltd. Integrated Camera]
isto@isto-ThinkPad-W510:~/uhubctl/uhubctl-master$

Turning off the power of all ports in Transcend, result is quiet ESD
isto@isto-ThinkPad-W510:~/uhubctl/uhubctl-master$ sudo ./uhubctl -l2-1.2 -a0
Current status for hub 2-1.2 [8564:4100 Transcend USB2.1 Hub, USB 2.10, 4
ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [0ab4:0010 esd electronic
system design gmbh (www.esd-electronics.com) USB2292 HJ005337]
Sent power off request
New status for hub 2-1.2 [8564:4100 Transcend USB2.1 Hub, USB 2.10, 4 ports]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0000 off
isto@isto-ThinkPad-W510:~/uhubctl/uhubctl-master$

>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
Hub in USB3 (blue connector in thinkpad)

isto@isto-ThinkPad-W510:~/uhubctl/uhubctl-master$ sudo ./uhubctl
Current status for hub 2-1 [8087:0020, USB 2.00, 8 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0103 power enable connect [17ef:1003 Lenovo Integrated Smart Card
Reader]
  Port 6: 0100 power
  Port 7: 0100 power
  Port 8: 0100 power
Current status for hub 4-2 [8564:4100 Transcend USB3.1 Hub, USB 3.10, 4
ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 4 [1d6b:0003 Linux 4.15.0-20-generic xhci-hcd xHCI
Host Controller 0000:0f:00.0, USB 3.00, 2 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 0263 power 5gbps U3 enable connect [8564:4100 Transcend USB3.1
Hub, USB 3.10, 4 ports]
Current status for hub 3-2 [8564:4100 Transcend USB2.1 Hub, USB 2.10, 4
ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [0ab4:0010 esd electronic
system design gmbh (www.esd-electronics.com) USB2292 HJ005337]
Current status for hub 3 [1d6b:0002 Linux 4.15.0-20-generic xhci-hcd xHCI
Host Controller 0000:0f:00.0, USB 2.00, 2 ports]
  Port 1: 0303 power lowspeed enable connect [046d:c050 Logitech USB-PS/2
Optical Mouse]
  Port 2: 0503 power highspeed enable connect [8564:4100 Transcend USB2.1
Hub, USB 2.10, 4 ports]
Current status for hub 1-1 [8087:0020, USB 2.00, 6 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0103 power enable connect [147e:2016 UPEK Biometric Coprocessor]
  Port 4: 0103 power enable connect [0a5c:217f Broadcom Corp Broadcom
Bluetooth Device 78DD08B2996C]
  Port 5: 0100 power
  Port 6: 0507 power highspeed suspend enable connect [17ef:480f Chicony
Electronics Co., Ltd. Integrated Camera]
isto@isto-ThinkPad-W510:~/uhubctl/uhubctl-master$

Turning off the power of all ports in Transcend, result is RUNNING ESD
isto@isto-ThinkPad-W510:~/uhubctl/uhubctl-master$ sudo ./uhubctl -l3-2 -a0
Current status for hub 4-2 [8564:4100 Transcend USB3.1 Hub, USB 3.10, 4
ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 4-2 [8564:4100 Transcend USB3.1 Hub, USB 3.10, 4 ports]
  Port 1: 00a0 5gbps Rx.Detect
  Port 2: 00a0 5gbps Rx.Detect
  Port 3: 00a0 5gbps Rx.Detect
  Port 4: 00a0 5gbps Rx.Detect
Current status for hub 3-2 [8564:4100 Transcend USB2.1 Hub, USB 2.10, 4
ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [0ab4:0010 esd electronic
system design gmbh (www.esd-electronics.com) USB2292 HJ005337]
Sent power off request
New status for hub 3-2 [8564:4100 Transcend USB2.1 Hub, USB 2.10, 4 ports]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0000 off
isto@isto-ThinkPad-W510:~/uhubctl/uhubctl-master$

mvp, 2018-11-02:
Thanks for the report, I added both to the table in a2b3695.
BTW, Transcend TS-HUB3K is the first hub I ever seen that identifies itself as USB 3.1.

closed 2018-11-02 by mvp