mvp/uhubctl #123: DeLock 87445 doesn't seem to be supported (but is on the list!)

vkahl, 2018-11-14:
I just bought a "DeLock 87445" because I wanted a small compatible hub that doesn't need an external power supply. The VID and PID are the same as in the list (05E3:0608). I tested it on three systems:
- Debian 9 (x86_64)
- Manjaro (x86_64)
- DietPi (Odroid C2, ARM64)

Unfortunately the output on all of them looks like this:
```bash
root@DietPi:~# uhubctl 
No compatible smart hubs detected!
Run with -h to get usage info.
```

So either I'm doing something wrong or this hub shouldn't be on the list.

Thanks and best regards!

mvp, 2018-11-14:
Removed Delock 87445 from the list in 465aa9b until proven otherwise.

closed 2018-11-14 by mvp

vkahl, 2018-11-14:
Thanks for the reply. That's unfortunate, I guess I have to get another one then. Good to know that the switching functionality is somehow related to the external power supply. The Delock does have a connector for one but it's optional...

FauthD, 2021-05-15:
Hi, 
this is very late, but just found about this site just now.
I had used the Delock 87445 with hub-ctrl back in ~2012. Works perfect.
Today I tested with uhubctl and it worked perfect.
Note: I used sudo since I did not create a udev rule yet. (Ubuntu 20-04, uhubctl -v = V2.1)

Perhaps there are different hardware version out there?
I recommend to list it with a warning about potential hardware revisions.

Greetings, Dieter Fauth

mvp, 2021-05-16:
@FauthD, can you please check if your hub supports turning VBUS off?
That is, connected phone stops charging when port is powered down. Thanks!

FauthD, 2021-05-17:
Hi, 
yes it turns off vbus. I have a bunch of these charge doctors (measure voltage and current).
I know others used this hub to control external devices with it.

mvp, 2021-05-18:
Thanks for your report @FauthD! Re-added this hub to supported list.

vkahl, 2021-05-19:
Well I guess there probably are different hardware versions then. After reading this I tried it again with mine but it still doesn't detect it as a compatible hub (uhubctl 2.3.0-1 on Debian 11/testing). Since I bought it many years later than FauthD, chances are that all newer variants won't work with uhubctl anymore. Or it's just some batches of them that don't work and I had bad luck.

mmilata, 2023-11-23:
Just bought one and it doesn't show up in uhubctl. It has ganged power switching according to lsusb: https://gist.github.com/mmilata/f22368fad0c86d14d9b22227a94f3b32