mvp/uhubctl #124: JSON output option

pabs3, 2018-11-15:
The uhubctl output currently requires custom parsing in order to use it in scripts. It would be nice to have an option for JSON output so that scripts can just use standard JSON parsing libraries to turn the output into a native data structure for their language.

BTW: I heard about uhubctl from this post:

https://joeyh.name/blog/entry/usb_drives_with_no_phantom_load/

mvp, 2018-11-15:
Thanks for submitting this feature request.
Do you have specific output format in mind? It would be helpful if you can propose one.
It should not be difficult to add `-j` (`--json`) option for json output, but we need to create stable and perhaps extensible output convention.

pabs3, 2018-11-15:
I was thinking of just a straight-forward transformation from the
current output to the equivalent JSON data structures.

So an array of hub objects, with attached metadata and each hub having
an array of port objects with attached metadata.

It is possible I've incorrectly inferred the internal data structures
from the output, but in any case I think the output should match the
internal structures.

Here is an example:

    [
        {
            "hub": "2-1",
            "usb": "2.00",
            "ports":
                [
                    {
                        "power": true,
                        "speed": "lowspeed",
                        "enabled": true,
                        "connected": true,
                        "vendor": 1118,
			"device": 1872,
			"description": "Microsoft Wired Keyboard 600",
                    },
                    ...
                ]
        },
        ...
    ]

-- 
bye,
pabs

https://bonedaddy.net/pabs3/

mvp, 2018-11-15:
Thanks for insight, I like the idea. I will try to implement this soon, but it will require quite a rewrite for all reporting functions. I t may take a while to do it, but at least it will justify making new official release :)

ezbik, 2023-05-02:
https://github.com/ezbik/scripts/tree/master/usbtree
https://github.com/tuna-f1sh/cyme
https://kellyjonbrazil.github.io/jc/docs/parsers/lsusb

benroeder, 2024-08-02:
working on this https://github.com/mvp/uhubctl/pull/575 , does this work for you ?

benroeder, 2024-08-02:
example output 
[example_json_output.json](https://github.com/user-attachments/files/16473013/example_json_output.json)

pabs3, 2024-08-03:
I'm no expert on the `uhubctl` internals or current output, but this proposed JSON output example looks good to me.

mvp, 2024-08-05:
Commented on #575, looks great but needs some fat trimming - both on json library used as well as size of json output.

benroeder, 2025-07-05:
ok I finally got round to doing this, sorry see https://github.com/mvp/uhubctl/pull/575