mvp/uhubctl #132: RPi 3B+

tjb8907, 2018-12-27:
I have this installed on my RPi 3B+, but can't seem to get it to work to switch off the usb ports. I run `sudo uhubctl -a 0` and i get:
`No compatible smart hubs detected!
Run with -h to get usage info.
There were permission problems while accessing USB.
To fix this, run this tool as root using 'sudo uhubctl',
or add one or more udev rules like below
to file '/etc/udev/rules.d/52-usb.rules':
SUBSYSTEM=="usb", ATTR{idVendor}=="2001", MODE="0666"
then run 'sudo udevadm trigger --attr-match=subsystem=usb'
`

I went through the steps to step the udev rules, and it works to run without sudo, but get the same message. Based on my readings, the RPi 3B should work, but has anyone used this  on a 3B+?

Trying to use it on my OctoPrint setup to stop the RPi from powering the printer board when the printer is off. If this doesnt work I can just just the power wire in my usb cable, but I would rather do it this way.

mvp, 2018-12-27:
RPi 3B+ has 2 hubs, daisy-chaned. You need to add 2 rules,.and make sure that they are activated by either rebooting or running udevadm command above. You may also check if it works by running uhubctl under sudo, which doesn't require udev rules.

mvp, 2018-12-27:
Also check issue #97

tjb8907, 2018-12-27:
No matter what I do, I can't get it spit out anything beside what I posted above..."No Compatible smart hubs detected", even when I use sudo. maybe I am missing another package that is required? I tried running 

`sudo apt-get install libusb-1.0-0-dev`

to make sure, and that came back saying the newest version was already installed. 

I do have a RPi 3 that is running Home Assistant on as well, I may try setting it up on that to see I can get it going on there first.

mvp, 2018-12-27:
Hmm, that doesn't sound good. Can you post output of `sudo lsusb -v`, without any extra USB devices connected? Specifically, I'm looking for wHubCharacteristic flags for onboard hubs.

tjb8907, 2018-12-28:
I believe I got it figured out. I was playing with some other ways to disable ports, and I think that was messing with it. Once I rebooted the Pi it worked, so it was either that, or just needed to reboot after installing uhubctl. It does work with sudo now, but I'm guessing I didn't do the udev right. I'm kind of a beginner when it comes to RPi stuff. I'll work on figuring it out and will post if I cant get it. 

Thanks for the help!

mvp, 2018-12-28:
Glad to hear it worked. I will close this then, please reopen if you think it requires additional action.

closed 2018-12-28 by mvp