mvp/uhubctl #137: TI TUSB4041PAPEVM

rshaub, 2019-02-05:
Writing to report the TI TUSB4041 evaluation module works with this library. http://www.ti.com/tool/TUSB4041PAPEVM

Output from uhubctl is as follows: Current status for hub 1-2 [0451:8142 1D000869EE46, USB 2.10, 4 ports]

mvp, 2019-02-05:
Thanks for your report! Your output seems to be trimmed.

Can you please try connecting phone to USB port, then turn it off using uhubctl and confirm that phone stops charging? Thanks!

rshaub, 2019-02-05:
I checked with one of my company's sensors, it definitely turns the power off to each port. I forgot to report the additional lines from usbctl. Here's the complete output:

Current status for hub 1-2 [0451:8142 1D000869EE46, USB 2.10, 4 ports]
    Port 1: 0100 power
    Port 2: 0100 power
    Port 3: 0100 power
    Port 4: 0100 power

OS is Ubuntu 16.04.5 LTS, if that matters.

mvp, 2019-02-05:
Awesome! I will add this hub to supported list today.

mvp, 2019-02-06:
Added TUSB4041PAPEVM in 2b88680d3b. Thanks for report!

closed 2019-02-06 by mvp