mvp/uhubctl #139: Does not work with Lenovo Internal HUB

xkill, 2019-02-18:
The following info from Lenovo P52

```
# uhubctl 
No compatible smart hubs detected!
Run with -h to get usage info.
```
```
# lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 10000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 8: Dev 3, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 8: Dev 3, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 9: Dev 4, If 0, Class=Vendor Specific Class, Driver=, 12M
    |__ Port 10: Dev 5, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 10: Dev 5, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 11: Dev 6, If 0, Class=Chip/SmartCard, Driver=, 12M
    |__ Port 12: Dev 7, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 12: Dev 7, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 14: Dev 8, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 14: Dev 8, If 1, Class=Wireless, Driver=btusb, 12M
```
```
# lsusb 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 06cb:009a Synaptics, Inc. 
Bus 001 Device 003: ID 04f2:b614 Chicony Electronics Co., Ltd 
Bus 001 Device 008: ID 8087:0aaa Intel Corp. 
Bus 001 Device 007: ID 04f2:b615 Chicony Electronics Co., Ltd 
Bus 001 Device 006: ID 058f:9540 Alcor Micro Corp. AU9540 Smartcard Reader
Bus 001 Device 005: ID 056a:5191 Wacom Co., Ltd 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

(I'll attach the lsusb -vvvv)

xkill, 2019-02-18:
[lshw-sanitize-numeric-notime.txt](https://github.com/mvp/uhubctl/files/2877379/lshw-sanitize-numeric-notime.txt)
[lshw-sanitize-numeric-notime-businfo.txt](https://github.com/mvp/uhubctl/files/2877380/lshw-sanitize-numeric-notime-businfo.txt)
[lsusb-t.txt](https://github.com/mvp/uhubctl/files/2877381/lsusb-t.txt)
[lsusb-vvvvv.txt](https://github.com/mvp/uhubctl/files/2877382/lsusb-vvvvv.txt)

mvp, 2019-02-18:
It just means that this hub doesn't advertise per port power switching support in it's descriptor.
Table of supported devices that we maintain lists all tested devices. However, there many hubs which are not included yet.
If you find device that works but not included, please report.

closed 2019-02-18 by mvp

roccopietrini, 2019-05-03:
Hi, just a quick question on this.. 1d6b:0003 is reported on the tested device list, but it doesn't seems to work to the user, and neither to me.. what that could mean? Thank you

mvp, 2019-05-03:
@roccopietrini, 1d6b:0003 is just vid:pid of original chipset manufacturer per http://www.linux-usb.org/usb.ids. In this case, 1d6b is Linux Foundation, 0003 is root USB 3.0 hub - so it's kind of catch-all generic id.

Hardware manufacturers are supposed to allocate new USB vendor ids from USB.org (and pay for it). Often, they don't want to do that and just use generic ids.