mvp/uhubctl #140: EspressoBin v5a internal USB3 HUB is supported (limited)

thiscantbeserious, 2019-02-23:
Good news:

The EspressoBin v5 SBC works with its internal USB3 HUB - tought it seems a bit bugged in that it will confuse USB3/2 from time to time (so output is unreliable) and you have to disable/reenable all 3 HUBs (with -a 2) together or it does nothing. 

The third HUB seems to be the debugging port - since it seems to do nothing for the USB2 hubs that shouldn't matter.

I'm Happy! Now I can turn my external USB HDD off after the backup is done programmatically.

Thanks for your very usefull utility!

mvp, 2019-02-23:
Thanks for report! USB 2/3 confusion may come from multiple hubs daisy chained. Can you please give uhubctl output for this hub? Make sure to use uhubctl built from master. Thanks!

thiscantbeserious, 2019-02-23:
I'm using build from today (master) anyway since its aarch64

```
$uname -a
Linux alarm 4.20.10-1-ARCH #1 SMP PREEMPT Fri Feb 15 18:12:49 MST 2019 aarch64 GNU/Linux
```

[lsusb -vvvv](https://pastebin.com/YR9BbQya)

```
$sudo uhubctl > uhubctl.log
$sudo uhubctl -a 0 -l 1 >> uhubctl.log
$sudo uhubctl -a 0 -l 2 >> uhubctl.log
#after here the disk was powered off
$sudo uhubctl -a 0 -l 3 >> uhubctl.log
$sudo uhubctl -a 2 -l 3 >> uhubctl.lpg
$sudo uhubctl -a 2 -l 2 >> uhubctl.log
$sudo uhubctl -a 2 -l 1 >> uhubctl.log
#here it was powered on again
```

[uhubctl.log](https://pastebin.com/02gcFhHJ)

mvp, 2019-02-23:
Hmm. This is really weird. I have never seen 1-port hubs before. I also have not observed root hubs supporting PPPS before. Perhaps it has something to do with with very recent Linux kernel 4.20 you are running.

There is no evidence of daisy chained hubs here, I only see 3 hubs with addresses 1, 2, and 3. As far as I can tell, they are all the same physical hub, 1 attached to ehci (old USB 2 interface), 3 is USB3 hub, and 2 is USB 2 compatibility dual partner for USB 3 hub.

Normally, you would only see one USB 2 dual partner for USB 3 hub. I believe solution for reliable control for this case is using 3 following commands in a row to turn off:

    uhubctl -a 0 -e -l 1
    uhubctl -a 0 -e -l 2
    uhubctl -a 0 -e -l 3

and same with -a 1 to turn back on.

mvp, 2019-02-24:
I have pushed commit https://github.com/mvp/uhubctl/commit/343e02b6345700e9adf736c1798b8c3f82ace355. It should improve USB2/3 handling, and should make it work better for your case.

Just using

    uhubctl -a 0 -l 3

should work correctly.
BTW, this device has following topology: bus `2` and `3` are representing one physical device (for USB2 and USB3 interfaces). But bus `1` is separate USB port (perhaps microUSB?), and it should be unrelated to your external HDD.

Can you please try building from master and confirm that this works for you? Thanks!

mvp, 2019-02-25:
Added to supported table in 1773ac260.

closed 2019-02-25 by mvp

thiscantbeserious, 2019-02-27:
uhubctl -a 0 -l 3

works of I trigger it twice - first one will disable usb3 second one usb2 then it works

mvp, 2019-02-27:
Can you please do few tests:

Run `uhubctl -a 0 -l 3` and provide output from it (perhaps twice, until power really goes off). Basically, I want to understand if switching off order of USB3 vs USB2 matters. If it doesn't turn off in one go, try resetting and doing it with `uhubctl -a 0 -l 3 -r 500`. -r is forcing to send turn off many times in quick succession - some devices need it because kernel may automatically turn device back on. Thanks!

thiscantbeserious, 2019-03-04:
uhubctl -a 0 -l 3 -r 500

Works, haven't had time for the other command since my main machine died :(

mvp, 2019-03-04:
Thanks, I guess it wasn't about the order. I'm not sure how to make it more robust against kernel automatically powering port back on... Ideas welcome on how this can be automated. I don't like using -r 500 for everything.

SlySven, 2019-03-16:
> Hmm. This is really weird. I have never seen 1-port hubs before. 

What about a USB active extension cable (to get past the 5m length limit for USB 2.0)? I've got one connected up to the RFID reader glued to the inside of the glass on my front door to *eventually* connect to the Home Automation RPi B+ so I can unlock the electric lock on the door with an RFID key/card - when I get around to finishing it. I would be curious to find out if it appears as a single port Hub with Power Switching... :thinking:

mvp, 2019-03-16:
If active extension cable worked as 1-port PPPS hub, that would be probably easiest to deploy and most popular application for uhubctl :) Please report if it works, @SlySven :)

SlySven, 2019-03-16:
:man_facepalming: I've just discovered that what I've got is an unpowered/passive extension - given that it is coupled to a RFID reader that appears normally as a USB 1.0/1.1 keyboard type device that is not an issue for my intended usage - but it does not help to prove/disprove the raised point here. :blush: 

OTOH the principle is probably sound as the whole reason for the active extensions is to overcome the timing limits imposed by the time it takes the signals to pass to and from the USB devices at each end and that means such a device *has to act as a **Hub*** device. So it would be well worth exploring...

thiscantbeserious, 2019-04-08:
Some further infos from my side 

`uhubctl -a 0 -l 3 -r 500` is the only command that really works reliably for me as it seems ... 

Anything less and it will sometimes result to uhubctl and lsusb thinking the ports are down while the device is still powered on for some reason. Nothing changed after I moved from Kernel 4.20 to >5 on the other hand ... so there's that.

I think I'll get a second EspressoBin for my parents with a different USB-HDD to have a cheap backup target to verify if everything is really working. I have yet to move to Armbian (currently on 4.19) to see if it's behaving any differently (still on Archlinux ARM).

mvp, 2019-04-08:
Check https://github.com/mvp/uhubctl/issues/90.
Basically what happens here is that some device drivers in kernel are surprised by USB device being turned off and automatically try to power it back on. When use option -r N, uhubctl will try to turn power off many times in quick succession, and it should suppress that as kernel driver will stop trying to re-enable. This may be eventually fixed in kernel, see more discussion here: https://www.spinics.net/lists/linux-usb/msg157413.html