mvp/uhubctl #142: Can't build on FreeBSD

rockyhotas, 2019-03-05:
Hello!
I'm running `FreeBSD 12.0-RELEASE` (amd64).
After cloning the repository, inside its directory I ran `make` and the following error is generated:

```
$ make
make: "/home/user/github/uhubctl/Makefile" line 20: Missing dependency operator
make: "/home/user/github/uhubctl/Makefile" line 22: Need an operator
make: "/home/user/github/uhubctl/Makefile" line 24: Missing dependency operator
make: "/home/user/github/uhubctl/Makefile" line 25: Missing dependency operator
make: "/home/user/github/uhubctl/Makefile" line 29: Need an operator
make: "/home/user/github/uhubctl/Makefile" line 31: Need an operator
make: "/home/user/github/uhubctl/Makefile" line 33: Missing dependency operator
make: "/home/user/github/uhubctl/Makefile" line 34: warning: duplicate script for target "ifeq" ignored
make: "/home/user/github/uhubctl/Makefile" line 21: warning: using previous script for "ifeq" defined here
make: "/home/user/github/uhubctl/Makefile" line 35: Need an operator
make: Fatal errors encountered -- cannot continue
make: stopped in /home/user/github/uhubctl
```

(This same repository, on Ubuntu 18.04, had instead been built very quickly).
Maybe it is a problem with `UNAME_S := $(shell uname -s)`?

mvp, 2019-03-05:
I think this is because FreeBSD make is different from the gnu make that is used commonly on Linux and Mac. Can you try building with gmake? I would love to find out make syntax that is compatible with both make flavors...

rockyhotas, 2019-03-05:
Oh, it may definitely be one of the reasons. Trying with `gmake`:

```
$ gmake
cc  -g -O0 -Wall -Wextra -std=c99 -pedantic -DPROGRAM_VERSION=\"v2.0.0-45-gc237\" uhubctl.c -o uhubctl -lusb
uhubctl.c:338:45: error: use of undeclared identifier 'LIBUSB_DT_SUPERSPEED_HUB'
    int desc_type = bcd_usb >= USB_SS_BCD ? LIBUSB_DT_SUPERSPEED_HUB
                                            ^
uhubctl.c:357:13: warning: implicitly declaring library function 'snprintf' with type 'int (char *, unsigned long, const
      char *, ...)' [-Wimplicit-function-declaration]
            snprintf(
            ^
uhubctl.c:357:13: note: include the header <stdio.h> or explicitly provide a declaration for 'snprintf'
uhubctl.c:534:21: warning: implicit declaration of function 'libusb_get_parent' is invalid in C99
      [-Wimplicit-function-declaration]
                if (libusb_get_parent(udev)      == dev &&
                    ^
uhubctl.c:534:50: warning: comparison between pointer and integer ('int' and 'struct libusb_device *')
                if (libusb_get_parent(udev)      == dev &&
                    ~~~~~~~~~~~~~~~~~~~~~~~      ^  ~~~
3 warnings and 1 error generated.
gmake: *** [Makefile:40: uhubctl] Error 1
```

mvp, 2019-03-05:
Thanks for trying this, much appreciated!

I wish I had remote access to FreeBSD system so I could fix all these issues...

1. Issue with `snprintf` seems similar to Darwin issue: this function is not defined in pure C more on FreeBSD, despite including stdio.h. Solution might be the same hack as `ifdef __APPLE__` block.
2. FreeBSD's libusb is not exactly the same as mainline libusb-1.0, and it looks like function `libusb_get_parent` either doesn't exist there, not defined or must be called under different name... Perhaps it's worth trying to link with mainline libusb?
3. This is induced error, not really relevant.

mvp, 2019-03-05:
Not defined `LIBUSB_DT_SUPERSPEED_HUB` means FreeBSD's libusb either doesn't support USB3, or this constant has different name.
Without this, uhubctl can't support USB 3.

I think case for using mainline libusb is getting stronger...

mvp, 2019-03-05:
I couldn't see libusb_get_parent in FreeBSD's libusb.h: https://github.com/freebsd/freebsd/blob/master/lib/libusb/libusb.h. There are some functions which might be used as a replacement, but I need to have access to working FreeBSD host to figure this out...

LIBUSB_DT_SUPERSPEED_HUB is not defined either, but I think we should be able to hack this with ifdef.

rockyhotas, 2019-03-05:
> Thanks for trying this, much appreciated!

Thanks to you for all your work!
 
> I wish I had remote access to FreeBSD system so I could fix all these issues...

AFAIK, something similar is provided by [sdf.org](https://sdf.org/), but for NetBSD.

>     1. Issue with `snprintf` seems similar to Darwin issue: this function is not defined in pure C more on FreeBSD, despite including stdio.h. Solution might be the same hack as `ifdef __APPLE__` block.

Ok! I modified `uhubctl.c` this way:

```
#if defined(__FreeBSD__) || defined(_WIN32)
int snprintf(char * __restrict __str, size_t __size, const char * __restrict __format, ...) __printflike(3, 4);
#include <libusb.h>
```
and so on; the error actually disappears:

```
$ gmake
cc  -g -O0 -Wall -Wextra -std=c99 -pedantic -DPROGRAM_VERSION=\"v2.0.0-45-gc237-dirty\" uhubctl.c -o uhubctl -lusb
uhubctl.c:339:45: error: use of undeclared identifier 'LIBUSB_DT_SUPERSPEED_HUB'
    int desc_type = bcd_usb >= USB_SS_BCD ? LIBUSB_DT_SUPERSPEED_HUB
                                            ^
uhubctl.c:535:21: warning: implicit declaration of function 'libusb_get_parent' is invalid in C99
      [-Wimplicit-function-declaration]
                if (libusb_get_parent(udev)      == dev &&
                    ^
uhubctl.c:535:50: warning: comparison between pointer and integer ('int' and 'struct libusb_device *')
                if (libusb_get_parent(udev)      == dev &&
                    ~~~~~~~~~~~~~~~~~~~~~~~      ^  ~~~
2 warnings and 1 error generated.
gmake: *** [Makefile:40: uhubctl] Error 1
```

rockyhotas, 2019-03-05:
> Not defined `LIBUSB_DT_SUPERSPEED_HUB` means FreeBSD's libusb either doesn't support USB3, or this constant has different name.
> Without this, uhubctl can't support USB 3.

In my case, this wouldn't be a problem: I don't have USB 3 sockets.

rockyhotas, 2019-03-05:
> I couldn't see libusb_get_parent in FreeBSD's libusb.h: https://github.com/freebsd/freebsd/blob/master/lib/libusb/libusb.h. There are some functions which might be used as a replacement, but I need [...]

Actually there are several packages available:

```
linux_libusb-11.0r261448_2     Linux-compatibility libusb
py27-libusb1-1.6.6             Pure-python wrapper for libusb-1.0
py36-libusb1-1.6.6             Pure-python wrapper for libusb-1.0
```

With `ifdef` I would not be able to figure out what to write (sorry, I am not so skilled with code). Just as a suggestion, why not to try a FreeBSD VM to run some tests?
And, if we do not succeed in building the code, maybe you could consider upgrading `README.md` with some information about possible FreeBSD issues. But first, I will help testing any other modification you think could be useful.

mvp, 2019-03-05:
Thanks for trying to fix snprintf problem. LIBUSB_DT_SUPERSPEED_HUB should be easy to bypass with ifdef as well.

Lack of libusb_get_parent seems most difficult problem. I am surprised that FreeBSD's libusb doesn't have it or any similar functions. Good news it has libusb_get_port_numbers, which actually should be enough to figure out full USB device topology. I will see if these is any way to change this code to avoid using libusb_get_parent - so it should work on all platforms...

mvp, 2019-03-05:
BTW, those packages are not going to solve this problem. Couple are just python wrappers, and linux_libusb is a wrapper to be able to run Linux executables linked to Linux's libusb. Which actually might make uhubctl compiled on Ubuntu work on FreeBSD :)

mvp, 2019-03-05:
Can you please try building from freebsd branch: https://github.com/mvp/uhubctl/tree/freebsd
I think this should solve all FreeBSD issues you encountered. Thanks!

rockyhotas, 2019-03-05:
Just tried and it all works! Only with `gmake`, however.
The executable is created and usable. Thank you so much!

mvp, 2019-03-05:
Thanks for testing this - I will merge this fix to master.
As for make build issue, supporting bsdmake is too much of an effort - bsdmake and gmake have quite different syntax. So, simple solution is use gmake to build on *BSD.
Thanks again!

mvp, 2019-03-05:
Fixed in https://github.com/mvp/uhubctl/commit/00d1a408b89bf0e7c549f7e45ed53efbbc6b931c.
Note that master is slightly different from freebsd branch - I found some other issues that were hidden (e.g. I did not check bus number before).

closed 2019-03-05 by mvp

mvp, 2019-03-05:
Added a note to README about gmake in https://github.com/mvp/uhubctl/commit/ae4be0a9d51fb6d844a4cfb0c182f5190dd74ff2.

rockyhotas, 2019-03-06:
Thanks to you so much.
No problem for `gmake` instead of bsdmake. And I'll keep referring to `freebsd` branch instead of `master`.

mvp, 2019-03-06:
Thanks. I have deleted freebsd branch, as all fixes from it are fully incorporated into master now.

rockyhotas, 2019-03-06:
Oh, ok, perfect!

SlySven, 2019-03-16:
As a Dual FreeBSD/Linux user I'd be interested in seeing how this fairs on the former... :grinning: