mvp/uhubctl #144: RPi B revision 1 works without issues

Piskvor, 2019-03-07:
Tested RPi B revision 1 - works, but is not listed in table yet (this is one of the oldest RPi models). There's an internal OTG USB hub managed by the SoC, with three ports:

```
# uhubctl
Current status for hub 1-1 [0424:9512, USB 2.00, 3 ports]
  Port 1: 0503 power highspeed enable connect [0424:ec00]
  Port 2: 0100 power
  Port 3: 0100 power
```

- Port 1 is the integrated ethernet adapter
- Ports 2 and 3 are exposed externally as USB 2.0 Type A connectors.

All three ports are controllable; re-enabling and/or reset restores normal functionality, even with the ethernet adapter.

#143

mvp, 2019-03-07:
Thanks for your report. Can you please check that ports 2 and 3 truly control power independently? On other RPi models, it may appear that ports are independently controlled (according to uhubctl), but in reality port 2 controls power for all ports 2-5. Trying to switch power for port 3+ won't really turn them off due to internal power wiring. Easiest way to check this to connect phone to USB and see if it is charging or not. Thanks!

Piskvor, 2019-03-08:
Yes. Each of the ports (1, 2, and 3) works independently of the others: switching any port on or off only affects that single port. Note that there is still voltage on the VCC and GND lines (dumb charging cable LED stays lit), but USB devices act as if physically unplugged. Switching the given port on restores functionality again.

mvp, 2019-03-08:
Well, inability to control VBUS is serious enough to not include device into compatibility list...
This seems to agree with info from Raspberry Pi developers https://github.com/mvp/uhubctl/issues/35#issuecomment-307875717, according to which only RPi 1B+ and higer support power switching.

Unless we have other information about Model B rev 1, I would not include it into supported list. We even have removed devices that do not turn off VBUS from the list before.

Piskvor, 2019-03-11:
Ok, that makes sense. Thanks for your time.

closed 2019-03-11 by Piskvor

mvp, 2019-03-11:
Thanks for your time too! If you happen to find any other compatible devices, please report them.