mvp/uhubctl #145: rock64 and QuantumZero QZ-HB05

dkimmortal, 2019-03-17:
hey,
have a [rock64](https://www.pine64.org) and a QuantumZero QZ-HB05 4 Port USB 3.0 Hub (powered)
which has a VIA VL813 Chip inside
when run:
sudo ./uhubctl
Current status for hub 5-1 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 0203 power 5gbps U0 enable connect [1058:25e2 Western Digital My Passport 25E2 575833314442363144354E55]
  Port 3: 0203 power 5gbps U0 enable connect [1058:25e2 Western Digital My Passport 25E2 57585331454337314B565932]
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 4-1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [0bc2:5070 Seagate FA GoFlex Desk 2HC015KJ]
Current status for hub 2 [1d6b:0002 Linux 4.4.167-1140-rockchip-ayufan-g6f266fb5d677 ehci_hcd EHCI Host Controller ff5c0000.usb, USB 2.00, 1 ports]
  Port 1: 0503 power highspeed enable connect [1058:259d Western Digital My Passport 259D 57584E31413536524B343855]
-----
First two passports connected to the QZ Hub, and the GoFLEX (though that is a USB 2.0 Powered Drive) and lastly the last mY Passport mentioned is connected outside of the hub onto a usb 2.0 port on the rock64. 
Why does it say Hub 5-1 and hub 4-1 instead of maybe hub '1a' and hub '1b'? as the second port on the rock64 is called hub 2. 
forgot to mention the QZ USB Hub is connected to the sole usb3 port on the rock64.

-Dk

mvp, 2019-03-17:
From README:

> Hub locations look like b-x.y.z, where b is USB bus number, and x, y, z... are port numbers for all hubs in chain, starting from root hub for a given USB bus.

Bus number and port numbers are part of physical USB topology. Note that Linux kernel is using exactly the same addressing method.

More important question here is: can you actually control power? If you connect say phone to USB port and turn off port with uhubctl, does phone stop charging?

mvp, 2019-03-17:
Regarding why hub is called just 2.
Yes, it does look odd. But this is what libusb (and presumably Linux USB stack) report.
It might be related to the fact that this odd hub is 1-port USB hub, which is weird in itself.
I have seen the same situation with EspressoBin - 1 port root hub is reported as having address just 3.

mvp, 2019-03-20:
@dkimmortal, can you please check if this device supports turning off VBUS?
Easiest way to check is connect a phone to USB port and try to turn port off using `uhubctl`.
If phone stops charging, I will add this device to compatibility list.
Thanks!

dkimmortal, 2019-03-20:
`sudo ./uhubctl
Current status for hub 5-1 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0080 off
  Port 2: 0203 power 5gbps U0 enable connect [1058:25e2 Western Digital My Passport 25E2 575833314442363144354E55]
  Port 3: 0203 power 5gbps U0 enable connect [1058:25e2 Western Digital My Passport 25E2 57585331454337314B565932]
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 4-1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0000 off
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [0bc2:5070 Seagate FA GoFlex Desk 2HC015KJ]
Current status for hub 2 [1d6b:0002 Linux 4.4.167-1140-rockchip-ayufan-g6f266fb5d677 ehci_hcd EHCI Host Controller ff5c0000.usb, USB 2.00, 1 ports]
  Port 1: 0503 power highspeed enable connect [1058:259d Western Digital My Passport 259D 57584E31413536524B343855]
`
hello, sorry for the late reply. Despite turning off power to port 1 where I plugged in a phone, it was still charging. So that would mean its not compatible right? 

I will check the ports of the rock64 itself now
same with the rock64, despite 'off' being shown in terminal; the phone is still powered

mvp, 2019-03-20:
Ugh, that's unfortunate...
Yes, in compatibility list I am trying to maintain only devices which truly turn power off.

That said, there was a time when I thought that few USB 3.0 hubs didn't work properly, only to realize later that I did not control them properly (I had to fix uhubctl to turn off power for both USB2 and USB3 virtual partner hubs).

I guess I will close this issue now. Please reopen if you find anything that proves that VBUS is turned off.

Thanks for your effort! Please report any other hubs if you find them work well :)

closed 2019-03-20 by mvp