mvp/uhubctl #146: Status check

jlnshr, 2019-03-21:
I am running ` ./uhubctl -a off -l 2-1.8 -p 1` to disable my port. Is there another way to check whether a port is activated or deactivated other than reading stdout and checking  for
 `New status for hub 2-1.8 [05e3:0608 USB2.0 Hub, USB 2.10, 4 ports]
Port 1: 0100 power` ?

mvp, 2019-03-21:
You can check the status by following hack:

    status=$(uhubctl -l 2-1.8 -p 1 | tail -1 | awk '{print $4}')

status will contain string `off` if port is off, and `power` if port is on.

mvp, 2019-03-21:
There is feature request to add JSON output which would make it easier to parse in scripts: https://github.com/mvp/uhubctl/issues/124.
But I have not found time to implement this yet, as it requires almost full rewrite of reporting functions.

closed 2019-03-21 by mvp