mvp/uhubctl #147: Attempt with Asus Z87A and "mute" USB hubs

rockyhotas, 2019-03-21:
Disclaimer: this is more a series of considerations (two motherboard tests) and questions, rather than a real issue.

When there are no compatible devices, `uhubctl` output is straightforward:

```
# ./uhubctl
No compatible smart hubs detected!
Run with -h to get usage info.
```

### Test 1

Asus Z87-PLUS is in the [Compatible USB hubs](https://github.com/mvp/uhubctl#compatible-usb-hubs). So, running Ubuntu 18.04 (` 4.15.0-46-generic`, `x86_64`) I checked the same-family motherboard Asus Z87A:

```
# ./uhubctl
Current status for hub 2-1 [8087:8000, USB 2.00, 8 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0100 power
  Port 6: 0100 power
  Port 7: 0100 power
  Port 8: 0100 power
Current status for hub 1-1 [8087:8008, USB 2.00, 6 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0100 power
  Port 6: 0100 power
```

Encouraged by the presence of an output, I tried to connect some a USB test key, but even powering off *all* the ports in hub `2-1` and `1-1`, the USB key still appear as full working (with both power and data).

```
# lsusb
Bus 002 Device 002: ID 8087:8000 Intel Corp.
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8008 Intel Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

This system has 4 USB buses: 1 USB 3.0, 3 USB 2.0. But only `Bus 003` and `Bus 004` seem to be accessible. In fact, a USB 2.0 key inserted in *any* physical port (USB 3.0 front panel, rear panel, USB 2.0 rear panel, motherboard internal 9-pin connectors with an adapter), always appears as a child of `Bus ''3`. I guess a USB 3.0 device would appear as a `Bus 004` child, even if I can't check this.

I thought `Bus 001` and `Bus 002` would be an OS abstraction, but they are listed also by NetBSD 8.0 on the same system:

```
# usbdevs
addr 0: xHCI Root Hub, vendor 8086
addr 0: xHCI Root Hub, vendor 8086
addr 1: EHCI root hub, vendor 8086
 addr 2: product 8008, Intel
addr 1: EHCI root hub, vendor 8086
 addr 2: product 8000, Intel
```

So, a conclusion and a question: while motherboard Asus Z87-PLUS is compatible, Asus Z87A is not; then, I wonder what exactly are these "mute" Buses `001` and `002` and why they are not related to any USB physical port.

Here, the `lsusb -v` output, where both the unusable Buses show `Per-port power switching` capability: [lsusb-v.txt](https://github.com/mvp/uhubctl/files/2991708/lsusb-v.txt).

### Test 2

With the same Ubuntu 18.04, I checked also motherboard Gigabyte Z97X-UD3H. All the results are identical to the Z87A, except that `Bus 001 Device 002` has `ID 8087:8009 Intel Corp.` and `Bus 002 Device 002` has `ID 8087:8001 Intel Corp.`. This motherboard is not compatible.

mvp, 2019-03-21:
```Current status for hub 2-1 [8087:8000, USB 2.00, 8 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0100 power
  Port 6: 0100 power
  Port 7: 0100 power
  Port 8: 0100 power
Current status for hub 1-1 [8087:8008, USB 2.00, 6 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0100 power
  Port 6: 0100 power
Current status for hub 4-3 [174c:3074 Asmedia ASM107x 00000000A001, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 3-3 [174c:2074 Asmedia ASM107x 00000000A001, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0303 power lowspeed enable connect [0e9c:0000 Streamzap, Inc. Streamzap Remote Control]

mvp, 2019-03-21:
As for "invisible" hubs on some motherboards, perhaps they represent additional USB headers which are often present on most motherboards - they are designed to connect extra USB headers to wire them to front of your (mini)tower computer. In many cases, these headers remain unconnected.

By the way, turning those hidden ports off will reduce power usage and should slightly reduce electric interference and perhaps increase audio quality, see this post https://bit.ly/2WhvEe6.

closed 2019-03-21 by mvp

rockyhotas, 2019-03-21:
> On my Asus Z87-PLUS, I see both Intel root hubs and AsMedia hubs.
> AsMedia is connected to backplane of this motherboard, and it does work to turn power off and on.
> E.g. I could turn off my Streamzap remote control as needed:

Maybe there is a different chip, even if it is still a Z87. It's great you can control it.

rockyhotas, 2019-03-21:
> As for "invisible" hubs on some motherboards, perhaps they represent additional USB headers which are often present on most motherboards - they are designed to connect extra USB headers to wire them to front of your (mini)tower computer. In many cases, these headers remain unconnected.

Yes, this is what I guessed, too. In fact, if you mean something like this:

[image]

(image from [anandtech](https://images.anandtech.com/doci/6970/ASUS%20Z87-C%20Top.jpg)), I purposedly bought an adapter like [this one](https://www.amazon.com/Motherboard-Female-Internal-Header-Adapter/dp/B01A917MHS), which can be plugged into the 9 pin motherboard connector. With a USB key connected to this adapter, powering off all the ports available in `uhubctl` does not produce any effect.

> By the way, turning those hidden ports off will reduce power usage and should slightly reduce electric interference and perhaps increase audio quality, see this post https://bit.ly/2WhvEe6.

Ok, thank you!

rockyhotas, 2019-04-04:
There are some updates. As already written, `uhubctl` in my motherboard can only use `hub 2-1 [8087:8000, USB 2.00, 8 ports]` and `hub 1-1 [8087:8008, USB 2.00, 6 ports]`. The other ones you have shown (which are the *right* ones!) are not available to me: maybe the two motherboards have a different hardware.

In my case, no device is attached to `hub 2-1` and `hub 1-1` because they are the legacy EHCI hubs, which are disabled when xHCI is used (as in default). If xHCI is disabled by BIOS instead, any USB peripherals will attach to `hub 2-1` and `hub 1-1`. `uhubctl` manages to turn down data connection (even if Linux immediately restores it, though I guess this can be avoided) but **not power**; the single USB Ports can also be Enabled/Disabled from BIOS, but (again) **not as regards power**. I think this is a hardware limitation. Exactly the same applies to Gigabyte Z97X-UD3H.

For anyone being curious, [this](https://rockyhotas.github.io/unix/2019/04/04/two-motherboards-and-usb-power-switching.html) is a blog post about these tests.

mvp, 2019-07-11:
@rockyhotas, reading your [blog post](https://rockyhotas.github.io/unix/2019/04/04/two-motherboards-and-usb-power-switching.html), it looks like you were able to compile and use uhubctl on NetBSD? Is that correct? Can you please tell what you needed to do to compile on NetBSD? Thanks!

rockyhotas, 2019-07-20:
> reading your [blog post](https://rockyhotas.github.io/unix/2019/04/04/two-motherboards-and-usb-power-switching.html)

First of all, thanks for checking it out, I'm glad!

> it looks like you were able to compile and use uhubctl on NetBSD? Is that correct? 

No. I used NetBSD only to compare with Linux the names/positions of the detected USB hubs.
I _tried_ to build `uhubctl` on NetBSD, but with no success.

>Can you please tell what you needed to do to compile on NetBSD? Thanks!

I installed `gmake` and `libusb1-1.0.22` from `pkgsrc` (using `pkgin install`). When I try to build, the error is:

```
$ gmake
cc  -g -O0 -Wall -Wextra -std=c99 -pedantic -DPROGRAM_VERSION=\"v2.0.0-68-g5f84\" uhubctl.c -o uhubctl 
uhubctl.c:35:31: fatal error: libusb-1.0/libusb.h: No such file or directory
compilation terminated.
gmake: *** [Makefile:43: uhubctl] Error 1
```

The same if I run this as `root`. Note that the base directory for installed packages in NetBSD is `/usr/pkg/`. In particular, `libusb1` includes the following files:

```
/usr/pkg/include/libusb-1.0/libusb.h
/usr/pkg/lib/libusb-1.0.la
/usr/pkg/lib/libusb-1.0.a
/usr/pkg/lib/libusb-1.0.so
/usr/pkg/lib/libusb-1.0.so.0
/usr/pkg/lib/libusb-1.0.so.0.1.0
/usr/pkg/lib/pkgconfig/libusb-1.0.pc
```
so maybe the `uhubctl` current `Makefile` does not look for the correct directory.

After a suggestion, I also tried to put

```
CFLAGS += $(shell pkg-config --cflags libusb)
LDFLAGS += $(shell pkg-config --libs libusb)
```
before line 40, `PROGRAM = uhubctl`. But this also generates the same error.
If you have any suggestions/ideas, I can perform further tests on NetBSD.

mvp, 2019-07-20:
Thanks a lot for checking this out.

Can you please try adding one more ifeq block in Makefile, like we did for Darwin:
https://github.com/mvp/uhubctl/blob/5f8426105c1a81663925cabfe9fc13002ae7f872/Makefile#L27-L32 Use proper CFLAGS and LDFLAGS per NetBSD convention.
Thanks!

rockyhotas, 2019-07-26:
> Thanks a lot for checking this out.

NP!

> Can you please try adding one more ifeq block in Makefile, like we did for Darwin:

This configuration works for me:

```
diff --git a/Makefile b/Makefile
index f16ac51..3a99bfa 100644
--- a/Makefile
+++ b/Makefile
@@ -33,6 +33,11 @@ endif
        LDFLAGS += -lusb-1.0
 endif
 
+ifeq ($(UNAME_S),NetBSD)
+        CFLAGS  += $(shell pkg-config --cflags libusb-1.0)
+        LDFLAGS += $(shell pkg-config --libs libusb-1.0)
+endif
+
 ifeq ($(UNAME_S),FreeBSD)
        LDFLAGS += -lusb
 endif
diff --git a/uhubctl.c b/uhubctl.c
index 78a2fac..f21804d 100644
--- a/uhubctl.c
+++ b/uhubctl.c
@@ -29,7 +29,7 @@
 #include <unistd.h>
 #endif
 
-#if defined(__FreeBSD__) || defined(_WIN32)
+#if defined(__FreeBSD__) || defined(__NetBSD__) || defined(_WIN32)
 #include <libusb.h>
 #else
 #include <libusb-1.0/libusb.h>
```
This way, `gmake` successfully creates the `uhubctl` executable:

```
$ gmake
cc  -g -O0 -Wall -Wextra -std=c99 -pedantic -DPROGRAM_VERSION=\"v2.0.0-71-g9cae-dirty\" -I/usr/pkg/include/libusb-1.0 uhubctl.c -o uhubctl -L/usr/pkg/lib -Wl,-R/usr/pkg/lib -lusb-1.0
```

There is some degree of freedom in choosing these parameters: one could alternatively use `include <libusb-1.0/libusb.h>` in `uhubctl.c` and set the `CFLAGS` and `LDFLAGS` differently in the `Makefile`, but I think the above solution is the best.
Given the above modifications, to build `uhubctl` with NetBSD, it is necessary to:

- install `libusb1` (`pkgin install libusb1`, which currently is `libusb1-1.0.22`)
- install `gmake` (`pkgin install gmake`)
- install `pkg-config` (`pkgin install gmake`)

mvp, 2019-07-26:
Awesome, thanks! I will add NetBSD as supported platform and consider using pkg- config everywhere.

mvp, 2019-07-27:
Added NetBSD support in 5db248771eb81921f0ec87a9075e7d741b8f88a2.