mvp/uhubctl #149: UH700: Shows Two Virtual Hubs? (TP-Link)

TheScarface, 2019-04-23:
**Hello! TP-Link UH700 is showing Two Virtual Hubs in the UHUBCTL.

What to do to solve this?**

`Current status for hub 1-1.1.2.1 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0503 power highspeed enable connect [05c6:f00e Qualcomm Mobile Router 0123456789ABCDEF]
  Port 2: 0503 power highspeed enable connect [0489:e0a6 Android Android 0123456789ABCDEF]
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-1.1.2 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0503 power highspeed enable connect [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [0489:e0a6 Android Android 0123456789ABCDEF]`

mvp, 2019-04-23:
Yes, this is proper output. This particular hub has 2 hubs daisy-chained: main 4-port hub with location(address) `1-1.1.2` has another 4-port hub attached to its port 1, thus having location `1-1.1.2.1`.

Because of daisy-chaining, 1 port is lost to internal connection, which makes it 7-port hub (4+4-1).

Note that TP-Link UH700 was previously included into compatibility list, but has since been removed, as we realized that VBUS (voltage) control is not supported on this hub. `uhubctl` willbe able to turn off data connection, but power will be still always connected.

closed 2019-04-23 by mvp

TheScarface, 2019-04-24:
Okay, Whenever I reset the whole hub, it throws the status of only single hub instead of both. Why is that?

Screenshot: https://i.imgur.com/MmkjUza.png

mvp, 2019-04-24:
I don't think you are telling the whole story here - I don't see which uhubctl command is actually executed by your script.

But, I can note 2 things:
1. TP-Link UH700 doesn't support VBUS control, which means that even though uhubctl does see hub reporting power off and on, actual power from attached devices is not revoked - connected phone may still continue to charge. Instead, I recommend getting any other hub from supported list.
2. For internally daisy-chained hubs like this one, you should never change power state for daisy-chaining USB port - one that connects slave hub to master hub. If you do, it will most likely completely confuse internal hub circuitry and will lead to undefined behavior. It seems like you are observing just that.