mvp/uhubctl #151: toggle script code won't run

The-Math-God, 2019-05-06:
hi i have made a toggle script code with uhubctl  

#!/bin/bash
if /home/pi/uhubctl/uhubctl -l 1-1.1 -p 3; then
   echo off
else
   echo on
fi

but i'm having an issue when i try to run it gives me a command not found error
sudo: /home/pi/scripts/toggle-printer: command not found
and the only command that is the problem is in the if section 
any ideas?

mvp, 2019-05-06:
It is hard to tell what is really wrong in your script - I think you only show part of it.

That said, can you install uhubctl as system tool in `/usr/sbin`? That should happen automatically if you do

    git clone https://github.com/mvp/uhubctl
    cd uhubctl
    make install
    sudo make install

Second, it would be really good idea to fix USB permissions as described here: https://github.com/mvp/uhubctl/blob/master/README.md#linux-usb-permissions
After doing it, you would not need to use sudo for uhubctl to work, and it will make your script work more reliably.

Also, if you want to get current port state, I recommend using this snippet:

    uhubctl -l 1-1.1 -p 3 | tail -1 | awk '{print $4}'

It will print `power` if power is on, and `off` is power is off.

closed 2019-05-06 by mvp

The-Math-God, 2019-05-06:
ok i did manage to find the issue for my previous code but no matter the status it always sent the first echo
now with your new code i can get status prints but i still have the problem but we are very close

heres my updated code
`#!/bin/bash
 if /home/pi/uhubctl/uhubctl -l 1-1.1 -p 3 | tail -1 | cut -f 6 -d " "; then
    echo on
else
    echo off
fi
`

mvp, 2019-05-07:
Just use this instead:

```
#!/bin/bash
uhubctl -l 1-1.1 -p 3 | tail -1 | cut -f 6 -d " "
```

It will print `power` or `off`.