mvp/uhubctl #153: Please add support for hub

xboxmasters, 2019-05-14:
Hello, I am using Eyeboot 49 port hub which consists of smaller 7 port hubs.
`Bus 008 Device 098: ID 1a40:0201 Terminus Technology Inc. FE 2.1 7-port Hub`

And uhubctl does not seem to recognize it, but I can confirm that it is working with hub-ctrl script.

Thank you.

mvp, 2019-05-14:
This hub does not advertise per-port power switching.
This is an example how lsusb output should look like for hub to be supported:
```
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             4
  wHubCharacteristic 0x00a9
    Per-port power switching
    Per-port overcurrent protection
    TT think time 16 FS bits
    Port indicators
  bPwrOn2PwrGood        1 * 2 milli seconds
  bHubContrCurrent    100 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0100 power
   Port 2: 0000.0100 power
   Port 3: 0000.0100 power
   Port 4: 0000.0303 lowspeed power enable connect
```
You hub advertises this instead:
```
  wHubCharacteristic 0x0088
    Ganged power switching
    Per-port overcurrent protection
```

Ganged power switching means hub only supports turning all 7 ports at once, not individually.
You need to find hub that supports it per-port.

xboxmasters, 2019-05-15:
Thank you for clarification, is it possible to do ganged switching at least with uhubctl then?

mvp, 2019-05-15:
Nothing is impossible. However, currently uhubctl is expecting per-port power switching across whole code. To support ganged switching, it's not enough to simply disable per-port check - e.g. it will have to understand that disabling 1 port will disable all of them. I'm not ready for this exercise yet.

closed 2019-05-15 by mvp

FD-, 2019-10-19:
@mvp Could you give any hints what would be needed to support ganged power switching? From the USB specs section 11.11,

> If a hub supports ganged power switching, then the power to all ports in a gang is turned on when any port in a gang receives a SetPortFeature(PORT_POWER) request. The power to a gang is not turned off unless all ports in a gang are in the Powered-off or Not Configured states.

AFAICT, this would mean issuing ClearPortFeature(PORT_POWER) on all ports should disable the power for the whole gang, right? Do the kernel and libusb support that as well?

mvp, 2019-10-19:
Theoretically yes, that should work. Practically it is very gray area.
E.g. there is no way to determine port gangs.
I could relax PPPS check to include ganged power switching hubs. But turning power on them would be hit or miss. E.g. there are 1-port ganged hubs, those definitely should be included as if they were per port.
But for 7 port hub, user would have to control all ports to be sure. I wouldn't be surprised if some ganged hubs report their descriptor incorrectly, and are in fact supporting per-port power switching instead.

FD-, 2019-10-19:
Thanks for your speedy response! I just want to hack something together that is specific to the hub chip I have here (which actually is the same as the OP mentions, the FE1.1s). So I just have to get rid of some PPPS checks in the hub descriptors? I assume these are at [Line 416](https://github.com/mvp/uhubctl/blob/743ecf226b72bbbadc28b05ef66ff0eabddd7ab3/uhubctl.c#L416) and [Line 689](https://github.com/mvp/uhubctl/blob/743ecf226b72bbbadc28b05ef66ff0eabddd7ab3/uhubctl.c#L689)?

mvp, 2019-10-19:
Yes, around line 416, add `|| lpsm == HUB_CHAR_COMMON_LPSM`.

FD-, 2019-10-19:
Thanks so much! It works (in that I'm able to observe a voltage change on the hub chip's PWRJ pin).

mvp, 2019-10-19:
Did you have to turn off all ports to remove voltage? Or hub behaves as if it actually supported per-port power switching?

FD-, 2019-10-19:
I just had to issue `sudo ./uhubctl -n 1a40 -a off`, assuming it would default to toggling all ports as per the `./uhubctl -h` information.

mvp, 2019-10-19:
Thanks. Can you try to control ports individually and check if that turns power per port?

mvp, 2019-10-19:
According to USB spec, if you turn off all ports for ganged hub, turning it back on for 1 port should turn on gang it belongs to. This would be a way to determine actual gangs. It could be all 7 ports. Or 2+2+2+1. Or 4+3 - only hub chip designer would know.

FD-, 2019-10-19:
Here's what I got:

```
pi@raspberrypi(ro):~/uhubctl$ sudo ./uhubctl -n 1a40 -a off -p 1
Current status for hub 1-1 [1a40:0101 USB2.0 HUB]
  Port 1: 0100 power
Sent power off request
New status for hub 1-1 [1a40:0101 USB2.0 HUB]
  Port 1: 0000 off
pi@raspberrypi(ro):~/uhubctl$ sudo ./uhubctl
Current status for hub 1-1 [1a40:0101 USB2.0 HUB]
  Port 1: 0000 off
  Port 2: 0103 power enable connect [0fe6:9700 USB 2.0 10/100M Ethernet Adaptor]
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1 [1d6b:0002 Linux 4.19.58+ dwc_otg_hcd DWC OTG Controller 20980000.usb]
  Port 1: 0503 power highspeed enable connect [1a40:0101 USB2.0 HUB]
pi@raspberrypi(ro):~/uhubctl$ sudo ./uhubctl -n 1a40 -a off -p 2
Current status for hub 1-1 [1a40:0101 USB2.0 HUB]
  Port 2: 0103 power enable connect [0fe6:9700 USB 2.0 10/100M Ethernet Adaptor]
Sent power off request
New status for hub 1-1 [1a40:0101 USB2.0 HUB]
  Port 2: 0000 off
pi@raspberrypi(ro):~/uhubctl$ sudo ./uhubctl -n 1a40 -a off -p 3
Current status for hub 1-1 [1a40:0101 USB2.0 HUB]
  Port 3: 0100 power
Sent power off request
New status for hub 1-1 [1a40:0101 USB2.0 HUB]
  Port 3: 0000 off
pi@raspberrypi(ro):~/uhubctl$ sudo ./uhubctl
Current status for hub 1-1 [1a40:0101 USB2.0 HUB]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0100 power
Current status for hub 1 [1d6b:0002 Linux 4.19.58+ dwc_otg_hcd DWC OTG Controller 20980000.usb]
  Port 1: 0503 power highspeed enable connect [1a40:0101 USB2.0 HUB]
pi@raspberrypi(ro):~/uhubctl$ sudo ./uhubctl -n 1a40 -a off -p 4
Current status for hub 1-1 [1a40:0101 USB2.0 HUB]
  Port 4: 0100 power
Sent power off request
New status for hub 1-1 [1a40:0101 USB2.0 HUB]
  Port 4: 0000 off
pi@raspberrypi(ro):~/uhubctl$ sudo ./uhubctl
Current status for hub 1-1 [1a40:0101 USB2.0 HUB]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0000 off
Current status for hub 1 [1d6b:0002 Linux 4.19.58+ dwc_otg_hcd DWC OTG Controller 20980000.usb]
  Port 1: 0503 power highspeed enable connect [1a40:0101 USB2.0 HUB]
pi@raspberrypi(ro):~/uhubctl$ sudo ./uhubctl -n 1a40 -a on -p 1
Current status for hub 1-1 [1a40:0101 USB2.0 HUB]
  Port 1: 0000 off
Sent power on request
New status for hub 1-1 [1a40:0101 USB2.0 HUB]
  Port 1: 0100 power
pi@raspberrypi(ro):~/uhubctl$ sudo ./uhubctl
Current status for hub 1-1 [1a40:0101 USB2.0 HUB]
  Port 1: 0100 power
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0000 off
Current status for hub 1 [1d6b:0002 Linux 4.19.58+ dwc_otg_hcd DWC OTG Controller 20980000.usb]
  Port 1: 0503 power highspeed enable connect [1a40:0101 USB2.0 HUB]
```

So the descriptor seems to reflect the individual port status. The PWRJ pin voltage changed (to high = 3.35V) as soon as all ports were off. Turning on a single port did change PWRJ to low again, but the descriptor only reflected the power change for just this port.

FD-, 2019-10-19:
A sane default would probably be to assume that hubs either have a single gang for all ports or individual per-port control. I know the specs allow more possibilities, but AFAICT, this would cover 99% of available end consumer hubs (at least those I came across while researching Raspberry Pi stuff).

mvp, 2020-12-09:
I have added support for switching power on ganged hubs 0cc09299fe. Please give it a try and report if it works on your hub