mvp/uhubctl #154: MOXA UPORT information

CWZ7605, 2019-06-14:
Dear Sir,
MOXA UPORT 407:
VID:110A
PID:0407
Release year: 2009
EOL: not yet

You can update these information. Thanks.

mvp, 2019-06-15:
Thanks for VID:PID info.
However, are you sure about 2009? I used 2017 because of this Amazon information: 
https://www.amazon.com/dp/B07B46JTW3
```
Date first listed on Amazon | November 15, 2017
```
I could not find any mentions of this hub before this date.
Even Moxa website does not have any documents dated earlier: https://www.moxa.com/en/support/product-support/software-and-documentation

CWZ7605, 2019-06-15:
Here is the Moxa China website for UPort 404/407 : https://www.moxa.com.cn/support/sarch_result.aspx?type=doc&prod_id=127&type_id=7
 The 1st edition manual was released on Jan 23, 2009.

Here is the Moxa website for UPort 404/407 : https://www.moxa.com/en/support/product-support/software-and-documentation/search?psid=56512
The 1st file "QIG for UPort 404/407 Series " was released on Jan 23, 2009.

So I think UPort 404/407 Series is realeased in 2009.

mvp, 2019-06-15:
Thanks, fixed!

closed 2019-06-15 by mvp