mvp/uhubctl #155: Thanks & some suggestion/questions.

toerless, 2019-06-15:
Thank you very much for this program.
 
1. I only started to play with it now. I tried it on a couple of hubs i have, but they all failed to actually cut power even though some of them looked as if they supported the API (uhubctl told that power was cut, even though it wasn't). With some of them, the connected 2.5" HDD LED actually turned off and it stopped spinning, so i thought it was working, and only a USB monitor showed me that there was still power applied to the disk, so the connected device was just put into some form of sleep mode To avoid other users having to repeat the experimentation, maybe a negative list of hubs tested that failed would be useful. If you want to start this, i will be happy to provide the names of the ones i tried via e.g.: a pull request or the like.

2. It seems to work fine with the AmazonBasics 4 port hub from your list. The port actually uses  2109:2811 for the USB2 part, but 2109:8110 for the USB3 part if you want to put that info into the README.md.

3. I want to use this for a bunch of 2.5" USB HDD for longer term media / backup storage, so only turn on a disk when needed. The main issue is the fact that after powering off the port, the  port, the device node for the disk stays, and when it is accessed, this leads to first I/O error and then the power to the port is reapplied (USB kernel driver ?), and the disk wakes up. That would be good to document. Also, if there is any way to effectively remove the /dev/nodes on power off (with some CLI) and reestablish them on poweron, that would be lovely. It seems that it may not be possible though without putting uhubctl into the kernel ?! *sigh*.

mvp, 2019-06-15:
Thanks for your feedback! To answer your questions:

1. Many hub manufacturers use proper USB chipsets, but fail to provide actual circuitry to cut off VBUS (5v power). Originally, I was filtering hubs by VID:PID I knew that definitely work. However, this approach makes it very difficult to collect data about new hardware that might work properly, so I removed this filtering. So now I can only recommend to use hubs from supported list.
Maintaining list of supported hubs inside uhubctl.c itself is not only expensive, it won't work: because many hub vendors don't actually change VID:PID from ones provided by chipset vendor. So it is possible to have two hubs with identical VID:PID, with one having working VBUS off, and another not having it.

2. Yes, we optionally list VID:PID in supported table just for reference (it is not used anywhere else). For USB3 hubs it could be either USB2 or USB3 part. We can list more, but frankly I don't think its really worth it - table is already quite wide.

3. This is a bug in Linux kernel - other platforms like Mac or FreeBSD don't have this issue.
Read more here: http://joeyh.name/blog/entry/usb_drives_with_no_phantom_load
and here: https://www.spinics.net/lists/linux-usb/msg157413.html
Perhaps one day this issue will be solved by Linux kernel developers, I don't know.

mvp, 2019-06-15:
Please reopen if you feel your questions were not addressed. Or open another issue if you have one. Thanks for your feedback!

closed 2019-06-15 by mvp

toerless, 2019-06-19:
Thanks you very much, mvp

Not sure if this issue tracker is the best place to do this type of Q&A, if not, please suggest another way, because i have not found one.

1. Maybe i didn't make myself clear in point 1: The positive list is great, but right now there will likely be a good amount of people trying to figure out if a specific other hub that is of interest would support uhubctl, but once they figure out the hub does not work, that investigation went nowhere, because the results of such testing are not tracked. Maybe just accept pull requests for testing results in a sub-directory where contributors could commit what they have tested (and what failed).

3. your answer to point 3 i quite interesting: Are you saying my experience with the device nodes would be better in freebsd than in linux, and if so, how ? I am trying to build me a file server with a lot of disks connected to USB hubs, but only activate the disk i need, and i'll probably put the whole thing into a VM anyhow, so it wouldn't matter which OS i use (except that i haven't used freebsd for 15 years now *sigh*). So if freebsd should be better with uhubctl, that could be a decisive argument for it. The other issue i have seen is that for me, linux reliably creates I/O errors on "higher" utilization. In my last tests, i could already see recurring I/O errors when just reading continuously from 3 or more USB 3 raw disk device nodes  connected to a single USB 3 hub (with different hubs). Worst case, USB MUST suck because its badly designed, but best case it's just linux that sucks, and i could overcome this with freebsd ;-)

mvp, 2019-06-20:
1. I have very simple approach to this - if someone is looking for working hardware, then anything in supported list is (mostly) guaranteed to work, and they should be looking to acquire specific hardware. If you have unlisted hardware that works, great - more power to you, but then please report back and we will add it to supported list. Good share of such reports is already coming as pull requests, which I usually quickly approve.

2. You should be able to repeat instructions at http://joeyh.name/blog/entry/usb_drives_with_no_phantom_load and make a solution that will work on Linux. 
  That article includes workarounds for dealing with non-disappearing device nodes.
I did not test FreeBSD very extensively. Few months ago there was a request to fix build on FreeBSD https://github.com/mvp/uhubctl/commit/00d1a408b89, so we know that it should be compilable and work well on that platform. Also note that it does work really well on MacOS, which is some respects is very similar to FreeBSD. 
  I would advise against using any VM with uhubctl - I do not believe any VM solution will properly propagate low-level USB control requests to turn power off/on from guest VM down to host kernel.
   Also, for USB storage consider using USB3 drives, hubs and cables, as only this will offer you acceptable performance. There are few supported USB3 hubs in the list, and they should work as expected.

toerless, 2019-06-20:
Thanks! 

Yes, i saw the freebsd thread. I'll certainly first test bare metal to check what works well. For VM i was planning to use PCI passthrough of one USB3 PCI controller. Already eliminated the option to get any type of useful paravirtualized USB behavior from any of the VM hosts. Target disks are USB3  2.5" HDDs for archival storage. 2109:0813

Alas, for the size constraints there is really only one known working USB3 hub small enough, the AmazonBasics. Thats a bit scary, i'll probably have to stock up on spares because that hub uses the EOL VL811 chip, so an update with a worse design could be coming any time *sigh*. Hence my attempts to also explore non listed hubs. Alas, even hubs that have the same USB IDs like supported hubs didn't work (Anker uses 2109:0813 like Plugable). I had a plugable too, but that one broke before i could test for uhubctl. Was also too large.

mvp, 2019-06-20:
Note that most USB chipsets properly support PPPS (per port power switching). However, device manufacturer can opt to disable it at chipset configuration level, and enable it only if appropriate power switching circuitry is actually present. Some manufacturers don't even bother doing that, so PPPS is advertised as available when it actually doesn't work.

Also, manufacturers are supposed to pay USB IF to acquire their own vendor id, and  change device vid:pid to some unique value. However, many don't want to spend $2000 for this and simply keep using chipset default vid, e.g. both AmazonBasics and Plugable use vid 2109, which belongs to chipset provider Via, Inc.

One more consideration: per USB 3.1 specification, any hub that has USB-C ports must support PPPS - it's not optional. You may consider looking for USB-C hubs, but so far I haven't found one.

toerless, 2019-06-22:
Using the 4 port amazon-basics in two tier setup:
motherboard->4phub0-{4phub1,4phub2,4phub3,4phub4}->12usb3disks

In this setup i can not make it work: when i switch off (with -e) the usb2 ports of e.g.: 4phub1, they come up again after e.g.: 30 seconds, or its not working at all. I can not first switch off the usb3 ports because then the disks fail over to usb2 and that creates even more undesired linux USB disk entries. Without -e i also get the duplicate disk entry issue (because it switches usb3 first off).

When i replace 4phub0 against a non-uhubctl compatible one (the ones i have to buy again because i had already returned it ;-), it works fine. I first switch off the usb2 ports on 4phub1...4 then i can port on/off disks by switching off/on the usb3 ports.

So i am fine, but was inquiring minds still wonders what happens.

mvp, 2019-06-22:
Don't rush returning Amazon Basics just yet. It is again Linux kernel issue - some kernel devices get surprised by lost power and try to enable themselves back on. You can solve this by using `-r N`, where N is number from 100 to 1000. Basically, it means uhubctl will send power off event many times in quick succession, and it will suppress device coming back on.

toerless, 2019-06-22:
I wasn't going to return these amazon basic hubs - i have no alternative. All i am doing is to replace 4phub0 with a non-amazon-basic. 
I can not follow your explanation. Let me rephrase my explanation:

motherboard---->non-uhubctl-usb3hub-(port1)->amazon-basic-usb3hub-(port1)->hdd

Start with everything switched on. switch off only usb2 port1 on the amazon-basic hub. Works great. hdd still running/unchanged because of usb3 still up. Then switch off usb3 port1 on the amazon-basic hup. Hdd switches off. Great!

Change non-uhubctl-usb3hub with another amazon-basic-usb3hub. Repeat experiment. This fails: usb2 on port 1 of amazon-basic hub connected to hdd goees down and comes up again.

Aka: i can not figure out why it would make a difference to have an uhubctl capable vs. non-capable one in this setup given how that i am not issuing any uhubctl commands to that hub and also given how the child amazon-basic-usb3hub will continue to draw power even if the HDD was switched off.
: when is tell 4phub1 to switch off port1 USB2, then the disk connected to that port will not loose power because USB3 is still up and running. But trying to do this will succeed

mvp, 2019-06-22:
Thanks for explanation. I'm not sure why do you want to daisy-chain 2 AmazonBasics hubs. But one possible explanation why it doesn't work is uhubctl may not reliably detect dual USB2/3 hubs to control and choose wrong one (I tried really hard to avoid this, but still possible with identical hubs). You should be able to control it explicitly with -e option.

If you are using -e and turning power off separately for USB2 then USB3 for 2nd hub in chain, it should work, unless there is some hardware issue with AmazonBasics hubs...

One more sanity check - make sure that all your hubs have own power supply. If they don't, results will be unpredictable.

toerless, 2019-06-23:
I need to daisy chain to get the number of ports i need. Only one available USB3 port on the motherboard. I just chained with the amazon basics just so i could stick with one product (sparing). Obviously, i do not need per-port power switching on the top hub, hence it just makes sparing a bit worse having to use two type of hubs.

The problem i was describing was with -e. Without -e its the same, except that without -e i would expect there to be possible timing issues anyhow.

This is inside a PC case, so all hubs have their power plug connected to the same main pc power supply 5A wired (20A). So the hubs are allways powered.

mvp, 2019-06-23:
Just a thought, if you want to use USB3 all the way, perhaps you can keep USB2 disabled? E.g. have a script that would periodically (every few minutes) turn off USB2 on ports where your hard drives are attached to? This way, it should reduce number of variables when you actually need to turn off your drives.

toerless, 2019-06-24:
The problem is jut that the USB2 hub will not stay "down" when i have the case of the two-tiered uhubctl capable hubs. But i am not bothering about this now given i have a workaround of using a non-uhubctl capable hub to connect all the uhubtl capable hubs.

The main issue is really that linux /dev nodes staying alive and then they are getting accessed and then the ports power up again. I am not sure though if i would call it a bug or a feature. As long as the linux kernel driver does not have a "power-off" feature natively, i would rather call it a feature. Especially because it is easily possible to first logically "remove" the disk, by writing to it's "remove" node in /sys file system, and then power down.

Only problem is figuring out which the right "remove" node in the /sys file system is. Many aliases, and little documentation. One can check what udisksd does, but better to understand why ;-)