mvp/uhubctl #157: DELL 1704FPT • The screen’s integrated USB hub works with uhubctl

mabam, 2019-06-16:
```
$ uhubctl
Current status for hub 20-9 [0424:a700, USB 2.00, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
```

mvp, 2019-06-17:
Thanks for your report!
Unfortunately, there are many device manufacturers who fail to include actual transistors to turn power off...

Can you please check this by connecting phone to USB port of your hub, then turn it off using uhubctl, and confirm that phone stops charging. Thanks!
Can you please confirm that power can be turned off? Easiest way to check is to connect a phone to USB port of this hub, turn it off using uhubctl and confirm that phone stops charging.

mabam, 2019-06-17:
I have a Wacom Tablet connected to it which I can successfully turn off and on using uhubctl. I have done so using port 1 and 2. Do you want me to test the remaining two ports as well?

mvp, 2019-06-17:
Awesome, that test should be enough. Best test would be connecting phone or USB lamp or fan and confirm that they stop charging, shining or spinning :)

I will add this device to supported list, thank you!

mvp, 2019-06-17:
Added in d8f7511.

closed 2019-06-17 by mvp

mabam, 2019-06-18:
> Awesome, that test should be enough. Best test would be connecting phone or USB lamp or fan and confirm that they stop charging, shining or spinning :)

The Wacom has a status light that stops shining when I turn off power.

I want to use uhubctl with a sleepwatcher script to restart the Wacom on Wake as it has a bug that results in the tablet staying off after a longer period of Sleep.

mabam, 2019-06-29:
FWIW:
I just figured that in my use case, when `sleepwatcher` calls `uhubctl` to shut down the Wacom tablet’s power and my machine goes to sleep, my machine immediately wakes up after entering sleep if the Wacom tablet was connected to the Dell’s USB 2 hub. If I connect the Wacom tablet to the AmazonBasics HU3641V1 USB 3 hub, I don’t have that problem and all works fine.

None of the two hubs are powered (I don’t use the power adaptor provided with the Amazon hub and the Dell doesn’t have one anyway). So apart from one being USB 2 and the other one USB 3, I don’t see any difference that might cause the issue.

mvp, 2019-06-29:
If hub doesn't have it's own power supply and is only powered by upstream USB port, I wouldn't trust if it advertised per port power switching support.

mabam, 2019-06-29:
I was assuming it wouldn't matter as long as the connected USB device doesn't consume more power than the hub can provide without a separate power supply. But I might be wrong here.

The AmazonBasics hub works fine without power supply in my case. And the Dell also does using uhubctl. Only when I use sleepwatcher to call uhubctl the machine won't stay asleep using the Dell screen's hub.

But it's not an issue for me as it does run with the Amazon hub. Just thought it is worth mentioning my findings here. You never know whom it might help.

Thank you for this tool, it does a great job!

mabam, 2020-05-23:
A _somewhat_ late addition:

> I have a Wacom Tablet connected to it which I can successfully turn off and on using uhubctl.

I do this to work around a bug in the Wacom Intuos S tablet which was never fixed by Wacom, causing it to not reconnect after wake unless it gets unplugged or powered off first.

In case this is of interest to you to add to your notable projects list, here’s how I did it:
https://translate.google.com/translate?hl=en&sl=de&tl=en&u=https%3A%2F%2Fwww.hackintosh-forum.de%2Fforum%2Fthread%2F42649-wacom-intuos-s-funktioniert-nicht-nach-dem-aufwachen%2F%3FpostID%3D514867%23post514867

It contains the XML for the macOS Launch Agent, which is messed up by Google translate. The XML code is best viewed in the untranslated original post: https://www.hackintosh-forum.de/forum/thread/42649-wacom-intuos-s-funktioniert-nicht-nach-dem-aufwachen/?postID=514867#post514867