mvp/uhubctl #158: uhubctl not recognizing Renesas uPD720201/2 as compatible

berndpfrommer, 2019-06-20:
I can't seem to get uhubctl to work on uPD720202. Would be great to have, so I can power cycle some USB3 cameras that seem to shut down randomly.

I know uPD720201 has not been reported as working yet, but uPD720202 is in the list of working hubs.

lspci | grep -i usb

04:00.0 USB controller: Renesas Technology Corp. uPD720201 USB 3.0 Host Controller (rev 03)
08:00.0 USB controller: Renesas Technology Corp. uPD720202 USB 3.0 Host Controller (rev 02)
09:00.0 USB controller: Renesas Technology Corp. uPD720202 USB 3.0 Host Controller (rev 02)
0a:00.0 USB controller: Renesas Technology Corp. uPD720202 USB 3.0 Host Controller (rev 02)
0b:00.0 USB controller: Renesas Technology Corp. uPD720202 USB 3.0 Host Controller (rev 02)

./uhubctl 
No compatible smart hubs detected!
./uhubctl --version
v2.0.0-63-gd8f7

System: 
Ubuntu 18.04LTS

Kernel:
kernel: 4.18.0-20-generic #21~18.04.1-Ubuntu SMP Wed May 8 08:43:37 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

mvp, 2019-06-20:
Can you please try running it under `sudo`?
Also, check if `sudo lsusb -v` shows per-port power switching support for your hub(s). If it does, uhubctl should work.

berndpfrommer, 2019-06-20:
All commands were run as root. Does the following output look valid to you? I know I have 3 cameras attached to one hub, and one camera to another one. But hubs are in fact built-in pci and pci-e cards.

Doesn't look like power switching is supported.

```
lsusb -v -d 1d6b:0003 | grep -i pow
      Self Powered
    MaxPower                0mA
    No power switching (usb 1.0)
   Port 1: 0000.0203 5Gbps power U0 enable connect
  Self Powered
      Self Powered
    MaxPower                0mA
    No power switching (usb 1.0)
   Port 1: 0000.0203 5Gbps power U0 enable connect
  Self Powered
      Self Powered
    MaxPower                0mA
    No power switching (usb 1.0)
   Port 1: 0000.0203 5Gbps power U0 enable connect
  Self Powered
      Self Powered
    MaxPower                0mA
    No power switching (usb 1.0)
   Port 1: 0000.02a0 5Gbps power Rx.Detect
  Self Powered
      Self Powered
    MaxPower                0mA
    No power switching (usb 1.0)
   Port 1: 0000.02a0 5Gbps power Rx.Detect
   Port 2: 0000.02a0 5Gbps power Rx.Detect
   Port 3: 0000.02a0 5Gbps power Rx.Detect
   Port 4: 0000.0203 5Gbps power U0 enable connect
  Self Powered
      Self Powered
    MaxPower                0mA
    No power switching (usb 1.0)
   Port 1: 0000.02a0 5Gbps power Rx.Detect
   Port 2: 0000.02a0 5Gbps power Rx.Detect
   Port 3: 0000.02a0 5Gbps power Rx.Detect
   Port 4: 0000.02a0 5Gbps power Rx.Detect
   Port 5: 0000.02a0 5Gbps power Rx.Detect
   Port 6: 0000.02a0 5Gbps power Rx.Detect
   Port 7: 0000.02a0 5Gbps power Rx.Detect
   Port 8: 0000.02a0 5Gbps power Rx.Detect
   Port 9: 0000.02a0 5Gbps power Rx.Detect
   Port 10: 0000.02a0 5Gbps power Rx.Detect
  Self Powered

```

mvp, 2019-06-20:
Most USB chipsets (including Renesas uPD7202x) support PPPS, but device manufacturer may opt to disable it at chipsets configuration level (perhaps because they don't provide actual circuitry to control power) - I believe this is what we have here. You can solve this by acquiring any other hub on supported list and attaching your devices via such hub.

closed 2019-06-20 by mvp

mvp, 2019-06-20:
For supported hubs, `sudo lsusb -v` would contain lines like these:

```
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             6
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
```