mvp/uhubctl #159: Still visible after disconnecting

jans23, 2019-06-21:
I use one of the D-Link USB hubs and experience that connected USB devices are still listed by Linux after disconnecting through uhubctl. Any idea?

mvp, 2019-06-21:
This is a bug in Linux kernel - other platforms like Mac or FreeBSD don't have this issue.
Read more here: http://joeyh.name/blog/entry/usb_drives_with_no_phantom_load
and here: https://www.spinics.net/lists/linux-usb/msg157413.html
Perhaps one day this issue will be solved by Linux kernel developers, I don't know.

closed 2019-06-21 by mvp

jans23, 2019-07-26:
@mvp Do you have a recommendation which device (USB hub or alike) could be used to fully disconnect and reconnect USB devices under Linux?

mvp, 2019-07-26:
Again, this is not hardware issue, but Linux kernel issue. So any hub listed in compatibility list should work. You can find workaround for this issue in article referenced in this FAQ: https://github.com/mvp/uhubctl/blob/master/README.md#usb-devices-are-not-removed-after-port-power-down-on-linux

szszszsz, 2019-07-27:
Hi @mvp! Thank you for developing the tool, and for the support!

Let me rephrase the question to: are you aware of any device or ready hardware solution, that would bypass the mentioned issue? 
Actually we face the same problem, as in [1] (we need the device to be powered down for some time, and removed from the `lsusb` list as well, then powered on). The way for removal using `...\remove` file works, but then there seems to be no way to reattach the device, which makes it unusable right now.
I think I have not checked the `udevadm` tool, though I believe this will not make the device disappear from the USB list.
  
One thing we are considering is using a custom hardware solution with a power or USB connection relay, to physically power-cycle the device instead of using power control of the hub itself (similarly to the physical removal of the device from an USB port). The other one is to adding a patch to the kernel for our needs, though this sounds more troublesome and less reliable, than the hardware solution (but would directly address the issue on the other hand).
It would be better to avoid doubling the work, and instead to find or buy a ready solution; especially, while the problem seems to be not that uncommon. 

Do you see any shortcomings in these ideas?

[1] https://www.spinics.net/lists/linux-usb/msg157413.html

mvp, 2019-07-27:
As I already said above, there is a workaround in FAQ, specifically [this comment](http://joeyh.name/blog/entry/usb_drives_with_no_phantom_load/#comment-1e4c9b2c34fdf66ddf3a333049881e82) from linked article - and yes, it is using udevadm to remove device file before or after power off.

Other solutions include patching kernel USB stack to properly detach devices upon receiving power off event. I do believe that this solution has to be implemented sooner or later.

Or, you can use other operating systems which don't have this problem, e.g. FreeBSD or NetBSD.

Really ugly solution could be using 2 daisy chained hubs. As far as I know, turning off power on master hub on port where second one is attached properly detaches all devices connected to slave hub.

szszszsz, 2019-07-31:
For the future reference: the expected effect was somewhat achieved by using:
```
sudo udevadm trigger --action=remove /sys/bus/usb/devices/${hub}.${port}/
```
After calling this command tools like `pkcs11-tool` were not seeing the device. However it was still visible in the `lsusb` list.

mvp, 2019-07-31:
Thanks! I have added explicit workaround in FAQ:
https://github.com/mvp/uhubctl/blob/master/README.md#usb-devices-are-not-removed-after-port-power-down-on-linux