mvp/uhubctl #160: Waiting on udev event for success

ptsneves, 2019-06-25:
Hello,

I am not sure this is an issue, nor if I am on track here.
I have a device which when turned on will expose an ethernet over usb interface. The problem is there seems to be some transience on this USB port.

My question is, would it be possible to only return success on a port operation after the bind event by the kernel is emitted? Something like is is emitted in `udevadm monitor``

`UDEV  [8720.504546] bind     /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.3/1-2.3.4 (usb)`

This way I think only when the kernel claims the usb device is up and drivers are loaded we are sure that we can use it.
Do you think this even makes sense? If it does I can try to implement it, given that there is already udev code in uhubctl.

mvp, 2019-06-26:
uhubctl doesn't really know anything about Linux kernel or udev. What makes you think it has any udev code? There is only suggestion about using udev to set up proper USB permissions, but that's about it.

All that uhubctl really does is to send low level USB control message to the hub to turn power off or on for a given port (provided that hub advertising per-port power switching). All these calls are done strictly using libusb API, and there is no Linux specific code anywhere.

If you want to wait on device appearance, isn't that what udev is for?

ptsneves, 2019-06-26:
> uhubctl doesn't really know anything about Linux kernel or udev. What makes you think it has any udev code? 

Well it objectively has udev code:
https://github.com/mvp/uhubctl/blob/d8f75116c84577af6b93f31a283c49372abd4727/uhubctl.c#L562
It looks like just for printing but indeed there is code and it is related to libusb function calls.

> If you want to wait on device appearance, isn't that what udev is for?
I want not a specific device connected to appear but that there is an event that the linux subsystem actually found a new usb port which it can use. The point is that uhubctl would not return before it is sure the port is up. I think I understand the architectural choice of just doing a single thing: Turn on the port, leaving the usability of the port as a separate matter.

mvp, 2019-06-26:
udev here just means USB device, lol. It has nothing to do with Linux udev subsystem.

mvp, 2019-06-27:
Closing this - not applicable. However, if you have an idea how to add Linux-specific extension, please feel free to submit pull request.

closed 2019-06-27 by mvp