mvp/uhubctl #162: can't make

SorenAndreasen, 2019-07-08:
I have make install on my RPI, but when I run make in the downloaded folder (after extraction), I get the following:

fatal: Not a git repository (or any of the parent directories): .git
cc  -g -O0 -Wall -Wextra -std=c99 -pedantic -DPROGRAM_VERSION=\"v2.0.0-dev\" uhubctl.c -o uhubctl -Wl,-z,relro -lusb-1.0

And after this, running "uhubctl" is not recognized as a valid command in that same directory

mvp, 2019-07-08:
This error should not happen if you use got clone https://github.com/mvp/uhubctl. Also, make sure to use `./uhubctl` to run it, or actually `sudo ./uhubctl`. You can avoid using sudo if you configure udev permissions, and avoid ./ if you install it using `sudo make install`.

SorenAndreasen, 2019-07-08:
okay that did it, thank you very much!

closed 2019-07-08 by SorenAndreasen