mvp/uhubctl #163: Support for Texas Instrument TUSB73x0

fsaraji, 2019-07-11:
It would be great to support this family of smart USB-hubs, since they are used more and more in Notebooks and industrial devices.

mvp, 2019-07-11:
If these hubs advertise PPPS (per-port power switching), they are already supported. But if they don't, well, it won't work. There is not much uhubctl itself can do if hardware lacks this capability.

If you find any particular hub that uhubctl would recognize and be able to control power, please report - we would be happy to add it to supported device list.

closed 2019-07-11 by mvp

fsaraji, 2019-07-11:
The hub supports PPPS. It showes up the list like this:

Current status for hub 8 [1d6b:0003 Linux 4.15.0-54-generic xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 7 [1d6b:0002 Linux 4.15.0-54-generic xhci-hcd xHCI Host Controller 0000:01:00.0, USB 2.00, 4 ports]
  Port 1: 0503 power highspeed enable connect [1e0e:9001 SimTech, Incorporated SimTech, Incorporated 0123456789ABCDEF]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power

Also the action on/off seem to work.... the only issue is that the ttyUSB port do not disappear... for the time the port is off, you cannot use them, but they still remain in /dev

any Idea?

fsaraji, 2019-07-11:
The controller is TUSB7340

mvp, 2019-07-11:
Can you try attaching a phone or USB lamp/fan to port if this hub, and check if phone stops charging, or USB lamp stops shining, or USB fan stops spinning?

Basically, I want to know if it can turn off VBUS (voltage on USB bus). Way too often, USB hubs just cut off data connection but don't turn off VBUS, and thus we don't include them into compatibility list.

I find it suspicious that USB2 counterpart of this USB3 hub advertises USB 2.00 - it should be USB 2.01 or 2.10 per USB3 spec.

mvp, 2019-07-11:
Regarding USB devices not disappearing from /dev, this is Linux kernel bug. You can find more why in issue https://github.com/mvp/uhubctl/issues/113.