mvp/uhubctl #165: Works with IBM ThinkPad X6 UltraBase

clavisound, 2019-07-22:
Output from uhubctl
```
uhubctl -a on -p 4
Current status for hub 1-4 [17ef:1000, USB 2.00, 4 ports]
  Port 4: 0000 off
Sent power on request
New status for hub 1-4 [17ef:1000, USB 2.00, 4 ports]
  Port 4: 0100 power

```
This is the docking station.
https://www.thinkwiki.org/wiki/ThinkPad_X6_UltraBase

And one that does *not* work although in lsusb -v is advertised as capable 'Per-port power switching'

This is does not work.
ID 05e3:0608 Genesys Logic, Inc. Hub

Labelled as apacer.

In reality it work like unbind. So if I have an iOS device is disconnected. But a fan continuous to work.

You are great! Thank you!

mvp, 2019-07-23:
Thanks for the update!

I tried to search this on Amazon, but there are just too many different models out there (all released in different years from 2006 to 2015), which is very confusing: 42W4635 (tablet style), 42W3107, 42W3108, 40Y8120, 42x4320 42x4321 and so on.

 Can you please provide model number printed on the back? I assume it should look similar to models I listed, e.g. 4NxNNNN. Thanks!

clavisound, 2019-07-23:
Sorry, I didn't knew about revisions! 42W3107 is the P/N. It's very old (in computer terms). Around 2006-2009
https://support.lenovo.com/gr/en/solutions/migr-62432

I thought you only needed those numbers: 17ef:1000.

Have a nice day.

mvp, 2019-07-24:
Added in 9cae5e2.

closed 2019-07-24 by mvp