mvp/uhubctl #166: smarthub not detected

coertklaver, 2019-07-25:
Hi,

my internal smarthub is not detected by uhubctl although it has the right wHubCharacteristic.

```
sudo lsusb:
Hub Descriptor:
  bLength              11
  bDescriptorType      41
  nNbrPorts             8
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood        0 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x00 0x00
  PortPwrCtrlMask    0xff 0xff
 Hub Port Status:
   Port 1: 0000.0100 power
   Port 2: 0000.0100 power
   Port 3: 0000.0100 power
   Port 4: 0000.0100 power
   Port 5: 0000.0100 power
   Port 6: 0000.0100 power
   Port 7: 0000.0100 power
   Port 8: 0000.0100 power
```

The FAQ states:
Note that uhubctl will only detect USB hubs which support per-port power switching. 

I am running 
```
uname -a:
Linux xxx 4.15.0-23-generic #25~16.04.1-Ubuntu SMP Fri May 25 04:50:20 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```

On a SuperMicro X10SRi-F with Xeon(R) CPU E5-2620 v4 @ 2.10GHz

Any idea why uhubctl does not see this hub as supported?

Thanks

mvp, 2019-07-25:
Please post output of the `sudo uhubctl` - it should detect your hub. Without sudo it would not be able to detect it, unless you fix USB permissions.

coertklaver, 2019-07-26:
Hello mvp,

Thanks for your fast response!

As per your request, I ran it as regular user:
```
.../uhubctl-master-master$ ./uhubctl
No compatible smart hubs detected!
Run with -h to get usage info.

```
And as root:
```

.../uhubctl-master-master$ sudo ./uhubctl
[sudo] password for xxxt: 
sudo: unable to execute ./uhubctl: Permission denied
Hangup
```
I assume I have to do something about permissions here, I followed the instructions under **Linux USB permissions**, but they don't seem to work for me. For instance the folder 

did not contain the file 52-usb.rules, so I created it.
Or should I just have adapted the file 52-digilent-usb.rules?

```
xxxx@xxx:~$ ls /etc/udev/rules.d/
52-digilent-usb.rules  52-usb.rules
xxxx@xxx:~$ cat /etc/udev/rules.d/52-usb.rules 
SUBSYSTEM=="usb", ATTR{idVendor}=="8087", MODE="0666"
SUBSYSTEM=="usb", ATTR{idVendor}=="1d6b", MODE="0666"
SUBSYSTEM=="usb", ATTR{idVendor}=="1dfa", MODE="0666"
xxxx@xxx:~$ 
xxxx@xxx:~$ cat /etc/udev/rules.d/52-digilent-usb.rules 
###########################################################################
#                                                                         #
#  52-digilent-usb.rules -- UDEV rules for Digilent USB Devices           #
#                                                                         #
###########################################################################
#  Author: MTA                                                            #
#  Copyright 2010 Digilent Inc.                                           #
###########################################################################
#  File Description:                                                      #
#                                                                         #
#  This file contains the rules used by UDEV when creating entries for    #
#  Digilent USB devices. In order for Digilent's shared libraries and     #
#  applications to access these devices without root privalages it is     #
#  necessary for UDEV to create entries for which all users have read     #
#  and write permission.                                                  #
#                                                                         #
#  Usage:                                                                 #
#                                                                         #
#  Copy this file to "/etc/udev/rules.d/" and execute                     #
#  "/sbin/udevcontrol reload_rules" as root. This only needs to be done   #
#  immediately after installation. Each time you reboot your system the   #
#  rules are automatically loaded by UDEV.                                #
#                                                                         #
###########################################################################
#  Revision History:                                                      #
#                                                                         #
#  04/15/2010(MTA): created                                               #
#  02/28/2011(MTA): modified to support FTDI based devices                #
#  07/10/2012(MTA): modified to work with UDEV versions 098 or newer      #
#  04/19/2013(MTA): modified mode assignment to use ":=" insetead of "="  #
#       so that our permission settings can't be overwritten by other     #
#       rules files                                                       #
#  07/28/2014(MTA): changed default application path                      #
#                                                                         #
###########################################################################

# Create "/dev" entries for Digilent device's with read and write
[... 93 lines cut ...]
   Port 6: 0000.0100 power
   Port 7: 0000.0100 power
   Port 8: 0000.0100 power
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 Unused
  bDeviceProtocol         0 Full speed (or root) hub
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0001
  Self Powered

```
In this 
Bus 002 Device 002: ID **8087**:8002 Intel Corp. 

and
wHubCharacteristic **0x0009**

seem to match. I'm lost.

Thanks for your time!

mvp, 2019-07-26:
That seems completely wrong, your sudo prohibits execution of uhubctl:
```
.../uhubctl-master-master$ sudo ./uhubctl
[sudo] password for xxxt: 
sudo: unable to execute ./uhubctl: Permission denied
Hangup
```
You should fix that first.

Your directory names look wrong as well.
Try this in empty directory:
```
git clone https://github.com/mvp/uhubctl
cd uhubctl
make
sudo ./uhubctl
```

coertklaver, 2019-07-26:
I seem to have gotten the wrong version from git. I first downloaded the master zip file, which gave me a different version then when cloning with git.

When cloning with git and running bot as user and su:
```

/uhubctl$ ./uhubctl
Current status for hub 2-1 [8087:8002]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0100 power
  Port 6: 0100 power
  Port 7: 0100 power
  Port 8: 0100 power
Current status for hub 1-1 [8087:800a]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0100 power
  Port 6: 0100 power
.../uhubctl$ sudo ./uhubctl
[sudo] password for xxxxx: 
sudo: unable to execute ./uhubctl: Permission denied
Hangup
```

mvp, 2019-07-26:
Glad it works for you now. Apparently udev changes are now working, so you don't need sudo. But your sudo config still seems out of whack.

closed 2019-07-26 by mvp

coertklaver, 2019-07-26:
I'll tweak with it a little more. Thanks for your help!

coertklaver, 2019-07-26:
It seems this device: 0424:2734 Microchip Tech USB2734
Is actually working. It sits in a Dell P4317Q monitor. You could add that one to the supported devices list.

However, when I switch off the whole hub:
```
 ./uhubctl -l 3-1 -a 0
Current status for hub 3-1 [0424:2734 Microchip Tech USB2734, USB 2.10, 5 ports]
  Port 1: 0303 power lowspeed enable connect [413c:2107]
  Port 2: 0303 power lowspeed enable connect [046d:c03d]
  Port 3: 0303 power lowspeed enable connect [413c:301a]
  Port 4: 0100 power
  Port 5: 0503 power highspeed enable connect [0424:274c Microchip Tech Hub Controller]
Sent power off request
New status for hub 3-1 [0424:2734 Microchip Tech USB2734, USB 2.10, 5 ports]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0000 off
  Port 5: 0000 off
```
I see the LED in a connected mouse go off, but after a few seconds it switches back on. Do you think the mouse driver is actively scanning USB port status and switching it back on? Because when after a few seconds I run:

```
./uhubctl -l 3-1
Current status for hub 3-1 [0424:2734 Microchip Tech USB2734, USB 2.10, 5 ports]
  Port 1: 0303 power lowspeed enable connect [413c:2107]
  Port 2: 0303 power lowspeed enable connect [046d:c03d]
  Port 3: 0303 power lowspeed enable connect [413c:301a]
  Port 4: 0000 off
  Port 5: 0000 off
```

ports 1-3 are 'automagically' on again.

Same happens with a USB memory stick (in port 3):

```
$ ./uhubctl -l 3-1
Current status for hub 3-1 [0424:2734 Microchip Tech USB2734, USB 2.10, 5 ports]
  Port 1: 0303 power lowspeed enable connect [413c:2107]
  Port 2: 0303 power lowspeed enable connect [046d:c03d]
  Port 3: 0503 power highspeed enable connect [08ec:0008]
  Port 4: 0000 off
  Port 5: 0000 off

 ./uhubctl -l 3-1 -p 3 -a 0
Current status for hub 3-1 [0424:2734 Microchip Tech USB2734, USB 2.10, 5 ports]
  Port 3: 0503 power highspeed enable connect [08ec:0008]
Sent power off request
New status for hub 3-1 [0424:2734 Microchip Tech USB2734, USB 2.10, 5 ports]
  Port 3: 0000 off
```

after a few seconds:

```
 ./uhubctl -l 3-1
Current status for hub 3-1 [0424:2734 Microchip Tech USB2734, USB 2.10, 5 ports]
  Port 1: 0303 power lowspeed enable connect [413c:2107]
  Port 2: 0303 power lowspeed enable connect [046d:c03d]
  Port 3: 0503 power highspeed enable connect [08ec:0008]
  Port 4: 0000 off
  Port 5: 0000 off

```

mvp, 2019-07-26:
This is FAQ entry:
http://github.com/mvp/uhubctl/blob/master/README.md#power-comes-back-on-after-few-seconds-on-linux

Try adding `-r 100` or higher.

coertklaver, 2019-07-26:
Sorry, case of RTFM.
That did the job!