mvp/uhubctl #167: F5U701cwBLK

faustri, 2019-08-07:
I bought the F5U701cwBLK from this page, as you can see it's called "BELKIN F5U701-BLK", it isn't found by uhubctl: https://deal.no/belkin-f5u701-blk/cat-p/c/p5543478

mvp, 2019-08-07:
Can you check it per this FAQ: https://github.com/mvp/uhubctl/blob/master/README.md#how-do-i-check-if-my-usb-hub-is-supported-by-uhubctl

If it still is not detected, it is possible that manufacturer has started using different USB chip in newer hub model.

faustri, 2019-08-07:
Yes, it's not listed there, I only get the onboard Intel NUC usbs (lsusb  says this is "Bus 002 Device 002: ID 8087:8001 Intel Corp."):

~/uhubctl ᛝ sudo ./uhubctl   
Current status for hub 2-1 [8087:8001]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0100 power
  Port 6: 0100 power
  Port 7: 0100 power
  Port 8: 0100 power

mvp, 2019-08-07:
Thats unfortunate. We observe this from time to time. E.g. DLink Dub-H7 rev A, then rev B and C do not, rev D starts working again.

I would have removed this hub from the list, but older models seem to actually work...
I guess will leave it as is...

closed 2019-08-07 by mvp